apache/nuttx #4037: Raw socket doesnot support non blocking and poll  and select  or any async read access

moataz-metwally, 2021-07-01:
I worked on a raw sockets and tried to use poll. after some debugging in deep in the network subsystem it turned out that raw socket neither support non blocking access nor poll.

acassis, 2021-07-01:
Hi @moataz-metwally should be nice if you could contribute some apps/testing/ sample to let it to be tested, do you have a minimum sample that works on Linux and fails on NuttX?

moataz-metwally, 2021-07-02:
Here is a sample code that works on linux but does not work on Nuttx btw there are many other issue with the net subsystem other than this issue but we can address it later.

```
#define ETH_SOCS_CNT 1
#define ETH_SOC_FD 0
#define POLL_TIMEOUT_MSEC   1
int socket_fd;
int ret = socket(AF_PACKET, SOCK_RAW | SOCK_NONBLOCK  , 0); 
if (ret == -1)
{
	printf("Cannon open ETH socket\n");
	
}
socket_fd = ret;

poll_fds[ETH_SOC_FD].fd = socket_fd;
poll_fds[ETH_SOC_FD].events = POLLIN | POLLPRI;

rc = ioctl(socket_fd, FIONBIO, (char*) &on); /* will fail with errno 88 */
if (rc < 0) {
	printf("ioctl() failed:%d\r\n",errno);
}

int events_num = poll(poll_fds, ETH_SOCS_CNT, POLL_TIMEOUT_MSEC);
printf("events_num:%x\r\n",events_num);

if (events_num < 0) {
	printf("Poll error: %d\r\n", errno); /* will fail with errno 88 */
}

```

You can go to net/socket/pkt_sockif.c, you will find that there is no any implementation for the poll and pkt_sockcaps.
the net subsystem as I see it uses pkt_sockcaps to get the flags set for the sockets, in case of raw socket , as you see, it returns zero so SOCK_NONBLOCK  flag is not taken into account.

Other issue with sendto, raw sockets doesnot support the use of sendto while on linux it works.

julianoes, 2023-02-22:
On that note I also found that `write` on socketcan will block if no CAN device is connected and - presumably - the buffer is full.

jmailloux, 2023-02-23:
+1.  I am trying to get socketCAN working on Pixhawk6C.  TX works.  I am having trouble on receive.  poll() doesn't seem to work as these other folks have said. 
Further to that, I can't seem to get candump working at all.  @julianoes, does candump work for you?

julianoes, 2023-02-23:
@jmailloux this might be a separate issue. For me, what happened was that the CAN message filters were not initialized to zero, so it was filtering stuff when it should not have.

The way to make it work - until I have figured out why it's not initialized at 0 - is to not use/allocate the filters like in [this workaround](https://github.com/PX4/NuttX/commit/b39310baea123ee27333e1743a80bbc7bd5c74e7).

I also asked in the ST community forums how the filters are initialized but didn't get an anser:
https://community.st.com/s/question/0D53W00001xRGuQSAW/how-is-can-message-ram-initializedzeroed-on-stm32h7

If you know anything about that I'm all ears.

jmailloux, 2023-02-26:
@julianoes, thanks. Your workaround fixed my issue as well.  poll now works, so does candump.
I have no idea why the filters are not initialized to 0.  I had thought the messages might be filtered, so I had this line to initialize the filters: setsockopt(CANmodule->fd, SOL_CAN_RAW, CAN_RAW_FILTER, NULL, 0);
That didn't seem to do anything.  BTW, seeing as we seem to be working on the same thing, perhaps we should keep in touch.  I don't see you on Dronecode Discord.  What would be the best way to keep in touch?
My end goal is to bring CanOpen support to PX4.  I've been using it with the NuttX character CAN device driver on Pixhawk 5.  I am now moving it to socketCAN on Pixhawk6C

julianoes, 2023-02-27:
Nice, that's good that you found the same and I'm not crazy.

> I don't see you on Dronecode Discord.

I'm there, I'll look out for you.

casaroli, 2026-11-08:
Fixed by #12995 and #17431.

closed 2026-11-08 by casaroli