apache/nuttx #4084: Stack overflow with apps on imxrt1064-evk:netnsh

fgiancane8, 2021-07-06:
Hello everyone,

I recently started hacking on a i.MX RT 1064 EVK board with NuttX (board support is not yet in any stable release, only on master branch) and I ported a very simple app running a Microsoft Azure IoT Hub Demo Example. (The library itself is inside Azure IoT Repo, for reference [here](https://github.com/Azure/azure-iot-sdk-c)

NuttX NSH shell is reporting an hard fault and the application is crashing after launched (I also step-by-step debugged it but apart from the crash there is really nothing wrong with the code...)

Please find attached here the trace:

```
nsh> Hello, from **** IoT Client Applicatioarm_hardfault: PANIC!!! Hard fault: 40000000
up_assert: Assertion failed at file:armv7-m/arm_hardfault.c line: 135 task: 
up_registerdump: R0: 20218b70 202102c8 f83a9851 bff9f07b 7004aee1 00000000 00000000 202199b8
up_registerdump: R8: 00000000 00000000 00000000 00000000 0000000d 202199b8 700250e9 700093e4
up_registerdump: xPSR: a1000000 BASEPRI: 00000080 CONTROL: 00000000
up_registerdump: EXC_RETURN: ffffffe9
up_dumpstate: sp:         20219828
up_dumpstate: stack base: 00000014
up_dumpstate: stack size: 04000000
up_dumpstate: ERROR: Stack pointer is not within the allocated stack
```

What is bugging me in this report is the stack pointer relative to stack base address and size. I understand that the panic is triggered because sp is outside stack bounded region, (00000014+ 04000000 is always outside 20200000).

In this board, on-chip RAM addresses start at 0x20200000 so, in principle, stack should be below those address and not starting from 0, to respect the alignment (and the size).

Do you think that there is something wrong in the linker script file on in the BSP for this particular board?
I can submit a PR if someone can hint me where to look for fixing the issue.

(Smaller standard programs like ping work fine, so I really do not understand if it is my executable or a system bug).

For reference, please find attached the `.config` file here

[config.txt](https://github.com/apache/incubator-nuttx/files/6771068/config.txt)

Any suggestion?

Thanks

acassis, 2021-07-06:
Hi @fgiancane8 you can start investing the Link Register (R14) and PC (R15) : 700250e9 700093e4
You can use this command: arm-none-eabi-addr2line -e nuttx 0x700250e9
Probably you need to increase the Stack size in the menuconfig: RTOS Features -> Stack and heap information

acassis, 2021-07-06:
Also please take a look here: https://cwiki.apache.org/confluence/display/NUTTX/ARMv7-M+Hardfaults%2C+SVCALL%2C+and+Debuggers

fgiancane8, 2021-07-07:
@acassis ,
Thank you for your inputs! 
I already checked the settings in RTOS Features -> Stack and heap information, and it seems to me that there is plenty of stack space allocated for the running tasks (also the ps command reports like 5 task running, so not much of space occupied).

I think the key here to understand the problem is the fact that the stack pointer points to an area in RAM while stack bounds (base and size) are in a totally different range , in which the stack pointer is not included. That is the reason why the `up_dumpstate: ERROR: Stack pointer is not within the allocated stack` error is raised.

Can you give me some hints where to search for stack area declaration initialization?

davids5, 2021-07-07:
This may help you debug it. https://www.youtube.com/watch?v=KZkAM_PVOi0 start at 17:25