apache/nuttx #4181: TCP iob_free.c assert

anchao, 2021-07-19:
Recently, we rebase of nuttx mainline to our internal branch. When running the HTTP audio streaming testing suite, we found that the audio streaming often crash in the iob_free.c:

assert:
`[  143.635905] [28] ap: up_assert: Assertion failed at file:iob/iob_free.c line: 101 task: media_src_movie`

backtrace:

```
(gdb) bt
#0  _assert (
    filename=0xef3f8000 "ﾭ\336ﾭ\336ﾭ\336ﾭ\336ﾭ\336ﾭ\336ﾭ\336ﾭ\336ﾭ\336ﾭ\336ﾭ\336ﾭ\336ﾭ\336ﾭ\336ﾭ\336ﾭ\336ﾭ\336ﾭ\336ﾭ\336ﾭ\336ﾭ\336ﾭ\336ﾭ\336ﾭ\336ﾭ\336ﾭ\336ﾭ\336ﾭ\336ﾭ\336ﾭ\336ﾭ\336ﾭ\336ﾭ\336ﾭ\336ﾭ\336ﾭ\336ﾭ\336ﾭ\336ﾭ\336ﾭ\336ﾭ\336ﾭ\336ﾭ\336ﾭ\336ﾭ\336ﾭ\336ﾭ\336ﾭ\336ﾭ\336", <incomplete sequence \336>..., 
    linenum=0) at assert/lib_assert.c:35
#1  0x570ac655 in iob_free (iob=0x5a223350 <g_iob_pool+79664>, producerid=IOBUSER_NET_TCP_READAHEAD) at iob/iob_free.c:101
#2  0x5839724a in iob_trimhead (iob=0x5a223350 <g_iob_pool+79664>, trimlen=468, producerid=IOBUSER_NET_TCP_READAHEAD) at iob/iob_trimhead.c:103
#3  0x582edf45 in tcp_readahead (pstate=0xef3f9ab0) at tcp/tcp_recvfrom.c:284
#4  0x582ef580 in psock_tcp_recvfrom (psock=0xf4804db0, buf=0xef3f9e90, len=1024, flags=0, from=0x0, fromlen=0xef3f9c94) at tcp/tcp_recvfrom.c:651
#5  0x582dc391 in inet_recvmsg (psock=0xf4804db0, msg=0xef3f9c90, flags=0) at inet/inet_sockif.c:1602
#6  0x56f404fe in psock_recvmsg (psock=0xf4804db0, msg=0xef3f9c90, flags=0) at socket/recvmsg.c:103
#7  0x56f3eded in psock_recvfrom (psock=0xf4804db0, buf=0xef3f9e90, len=1024, flags=0, from=0x0, fromlen=0x0) at socket/recvfrom.c:92
#8  0x56f3efc1 in nx_recvfrom (sockfd=6, buf=0xef3f9e90, len=1024, flags=0, from=0x0, fromlen=0x0) at socket/recvfrom.c:139
#9  0x56f3f03b in recvfrom (sockfd=6, buf=0xef3f9e90, len=1024, flags=0, from=0x0, fromlen=0x0) at socket/recvfrom.c:204
#10 0x56f3e9ee in recv (sockfd=6, buf=0xef3f9e90, len=1024, flags=0) at socket/recv.c:61
#11 0x5812922f in tcp_read (h=0xf2201d00, buf=0xf4a05208 "\027\003\003@", size=5) at libavformat/tcp.c:267
#12 0x58105759 in retry_transfer_wrapper (h=0xf2201d00, buf=0xf4a05208 "\027\003\003@", size=5, size_min=1, transfer_func=0x58128f55 <tcp_read>) at libavformat/avio.c:375
#13 0x58105d8c in ffurl_read (h=0xf2201d00, buf=0xf4a05208 "\027\003\003@", size=5) at libavformat/avio.c:410
#14 0x5812b048 in mbedtls_recv (ctx=0xf2201d00, buf=0xf4a05208 "\027\003\003@", len=5) at libavformat/tls_mbedtls.c:102
#15 0x57e80cb2 in mbedtls_ssl_fetch_input (ssl=0xf2001f6c, nb_want=5) at library/ssl_msg.c:2033
#16 0x57e953fa in ssl_get_next_record (ssl=0xf2001f6c) at library/ssl_msg.c:4536
#17 0x57e90d71 in mbedtls_ssl_read_record (ssl=0xf2001f6c, update_hs_digest=1) at library/ssl_msg.c:3898
#18 0x57e9b905 in mbedtls_ssl_read (ssl=0xf2001f6c, buf=0xf2a8c400 "\215\n\005\022]a\210\017\a!K\032\063\032N\247\335C\255", <incomplete sequence \366>, len=32768) at library/ssl_msg.c:5363
#19 0x5812cc46 in tls_read (h=0xf2201e00, buf=0xf2a8c400 "\215\n\005\022]a\210\017\a!K\032\063\032N\247\335C\255", <incomplete sequence \366>, size=32768) at libavformat/tls_mbedtls.c:304
#20 0x58105759 in retry_transfer_wrapper (h=0xf2201e00, buf=0xf2a8c400 "\215\n\005\022]a\210\017\a!K\032\063\032N\247\335C\255", <incomplete sequence \366>, size=32768, size_min=1, 
    transfer_func=0x5812cb29 <tls_read>) at libavformat/avio.c:375
#21 0x58105d8c in ffurl_read (h=0xf2201e00, buf=0xf2a8c400 "\215\n\005\022]a\210\017\a!K\032\063\032N\247\335C\255", <incomplete sequence \366>, size=32768) at libavformat/avio.c:410
#22 0x5790de66 in http_buf_read (h=0xf1e010c0, buf=0xf2a8c400 "\215\n\005\022]a\210\017\a!K\032\063\032N\247\335C\255", <incomplete sequence \366>, size=32768) at libavformat/http.c:1437
#23 0x5790eb6f in http_read_stream (h=0xf1e010c0, buf=0xf2a8c400 "\215\n\005\022]a\210\017\a!K\032\063\032N\247\335C\255", <incomplete sequence \366>, size=32768) at libavformat/http.c:1511
#24 0x5791085a in http_read (h=0xf1e010c0, buf=0xf2a8c400 "\215\n\005\022]a\210\017\a!K\032\063\032N\247\335C\255", <incomplete sequence \366>, size=32768) at libavformat/http.c:1635
#25 0x58105759 in retry_transfer_wrapper (h=0xf1e010c0, buf=0xf2a8c400 "\215\n\005\022]a\210\017\a!K\032\063\032N\247\335C\255", <incomplete sequence \366>, size=32768, size_min=1, 
    transfer_func=0x57910682 <http_read>) at libavformat/avio.c:375
#26 0x58105d8c in ffurl_read (h=0xf1e010c0, buf=0xf2a8c400 "\215\n\005\022]a\210\017\a!K\032\063\032N\247\335C\255", <incomplete sequence \366>, size=32768) at libavformat/avio.c:410
#27 0x578a65c9 in read_packet_wrapper (s=0xf3c01a80, buf=0xf2a8c400 "\215\n\005\022]a\210\017\a!K\032\063\032N\247\335C\255", <incomplete sequence \366>, size=32768) at libavformat/aviobuf.c:513
#28 0x578a7a0f in fill_buffer (s=0xf3c01a80) at libavformat/aviobuf.c:562
#29 0x578a9763 in avio_read (s=0xf3c01a80, buf=0xf362866e '\276' <repeats 200 times>..., size=786) at libavformat/aviobuf.c:655
#30 0x57aa67a5 in append_packet_chunked (s=0xf3c01a80, pkt=0xef3fb590, size=1024) at libavformat/utils.c:288
#31 0x57aa6cdd in av_get_packet (s=0xf3c01a80, pkt=0xef3fb590, size=1024) at libavformat/utils.c:311
#32 0x579e48e7 in mp3_read_packet (s=0xf3002a80, pkt=0xef3fb590) at libavformat/mp3dec.c:454
#33 0x57ab5585 in ff_read_packet (s=0xf3002a80, pkt=0xef3fb590) at libavformat/utils.c:869
#34 0x57ad194b in read_frame_internal (s=0xf3002a80, pkt=0xef3fb590) at libavformat/utils.c:1600
#35 0x57ad9356 in av_read_frame (s=0xf3002a80, pkt=0xef3fb590) at libavformat/utils.c:1802
#36 0x580de245 in amovie_get_frame (ctx=0xf505fd40, oframe=0xef3fb6b0) at libavfilter/src_movie_ext.c:460
#37 0x580e0809 in amovie_thread (arg=0xf505fd40) at libavfilter/src_movie_ext.c:613
#38 0x570839bd in pthread_startup (entry=0x580e055d <amovie_thread>, arg=0xf505fd40) at pthread/pthread_create.c:59
#39 0x5837eaf7 in pthread_start () at pthread/pthread_create.c:191
#40 0xdeadbeef in ?? ()
```
value:

```
(gdb) p *next
$2 = {io_flink = 0x5a2228d8 <g_iob_pool+76600>, io_len = 1518, io_offset = 0, io_pktlen = 1256,
  io_data = "\003\233\313Ac\203\003\316\b\324\027\211\233\365\320\061\201\304\021{|\230\331\344\035iMr\224[\312H\314\031\247}}\257JQ\206\005\251\061t\235\235,\241\035\240\021،\217ah\254AW\032T\352O\360\2
60cR\372\066>E\306\066\063\")L\261*]\207\237%\265!\203\003\030)\353To\303!\005\032CbZ\fO\035\207\227;\264\251\240r\260\220\303>\220\025ኤm\317_\217ơ\016T\221\360^b\365<\b\017.\300\244١\262\v\227Z\372\272A
\177A\304\345\300\016`\022\364Ѷ\034\253\301\357\031\250\376\216\242\261\241u\345[\253\027\rG\227\a\222v@\004y!\233\215\377%b\304\070+Wΰ\374\363@\204\227P@\035\357\327\065A"...}
```

iob config:

```
 982 CONFIG_MM_IOB=y
 983 CONFIG_IOB_NBUFFERS=72
 984 CONFIG_IOB_BUFSIZE=1518
 985 CONFIG_IOB_NCHAINS=72
 986 CONFIG_IOB_THROTTLE=8
 987 CONFIG_IOB_NOTIFIER=y
 988 CONFIG_IOB_NOTIFIER_DIV=4
```

However, there is no such assert before iob optimization:

```
commit 4878b7729cfb21f6dd7a0eaf6166e39a07be6e11
Author: YAMAMOTO Takashi <yamamoto@midokura.com>
Date:   Fri Jun 18 14:34:21 2021 +0900

    tcp: simplify readahead
    
    Do not bother to preserve segment boundaries in the tcp
    readahead queues.
    
    * Avoid wasting the tail IOB space for each segments.
      Instead, pack the newly received data into the tail space
      of the last IOB. Also, advertise the tail space as
      a part of the window.
    
    * Use IOB chain directly. Eliminate IOB queue overhead.
    
    * Allow to accept only a part of a segment.
    
    * This change improves the memory efficiency.
      And probably more importantly, allows less-confusing
      recv window advertisement behavior.
      Previously, even when we advertise N bytes window,
      we often couldn't actually accept N bytes. Depending on
      the segment sizes and IOB configurations, it was causing
      segment drops.
      Also, the previous code was moving the right edge of the
      window back and forth too often, even when nothing in
      the system was competing on the IOBs. Shrinking the
      window that way is a kinda well known recipe to confuse
      the peer stack.
```

@yamt san, Could you have a look?

yamt, 2021-07-20:
i will take a look.

yamt, 2021-07-20:
@anchao i think https://github.com/apache/incubator-nuttx/pull/4185 fixes this issue.

GUIDINGLI, 2021-07-20:
Test pass.
Confirm #4185 fix iob_free.c assert.

anchao, 2021-07-20:
@yamt , Thanks a lot for the quickly fix, I will close this PR!

closed 2021-07-20 by anchao