apache/nuttx #4188: nucleo_h743zi peripheral use

eden-desta, 2021-07-20:
Hello!

I am just starting out with NuttX and I am extremely stuck on trying to figure out how to access specific peripherals using my nucel h743zi board. 

For the purpose of background information I am able to get the simple `Hello` application to run using the nsh configuration. 
Now I am trying to run an example that uses any peripherals. The simplest one was the `leds` example that is provided in the `apps` folder. I used the `menuconfig` in order to add the application, and also added the LED drivers via `Device Drivers -> LED Support -> LED Driver -> Generic Lower Half LED Driver`.

I build via `make` and flash the code using `openocd` the `nsh` terminal shows up. And i run `leds` it returns an error
```
leds_main: Starting the led_daemon
leds_main: led_daemon started

led_daemon (pid# 6): Running
led_daemon: Opening /dev/userleds
led_daemon: ERROR: Failed to open /dev/userleds: 2
led_daemon: Terminating
```

I guess in short I am looking to understand how to configure peripherals for the board of my choice in order to be able to do anything that I want in particular. 

Apologies, there might just be something fundamental I missed while trying to get familiar with NuttX. Thank you in advance! Looking forward to hearing back from you!

michallenc, 2021-07-20:
Hello,

I think the problem is that you do not register the device /dev/userleds/ in board level code. If you take a look into this part of code for Teensy 4.1 board https://github.com/apache/incubator-nuttx/blob/master/boards/arm/imxrt/teensy-4.x/src/imxrt_appinit.c#L72, you can see that I call function userled_lower_initialize(LED_DRIVER_PATH) where LED_DRIVER_PATH is "/dev/userleds". That part of the code registers the device. 

If you write something like that into boards/arm/stm32h7/nucleo-h743zi/src/stm32_appinitialize.c, then it should do the trick. If you then run NuttX and execute command "ls dev/", you should see "userleds" there and the application should also be working. You also have to turn off CONFIG_ARCH_LEDS (this options use the led to show the status of NuttX - if it is turn on, NuttX is running, if the led is blinking then there was some problem with booting) in order to use user led.

I actually had the exactly same issue when I was beginning with NuttX. :)

Most of the NuttX peripherals have to be registered in board level section. It is usually done by calling function in architecture level and/or registering the device. If you take a look at this file https://github.com/apache/incubator-nuttx/blob/master/boards/arm/stm32h7/nucleo-h743zi/src/stm32_pwm.c, the pwm is initialized by calling stm32_pwminitialize() which is defined in arch/arm/src/stm32h7/stm32_pwm.c and then the device is register by pwm_register("/dev/pwm0", pwm) which is defined in drivers/timers/pwm.c. The same process is usually done with other peripherals (CAN, encoder, I2C and so on).

I hope this helps.

saramonteiro, 2021-07-20:
From what I see here, nucleo-h743zi board userleds implementation is incomplete.
We already have `nuttx/boards/arm/stm32h7/nucleo-h743zi/src/stm32_userleds.c`, so what is missing is to register it using `userled_lower_initialize("/dev/userleds");`.

Altough @michallenc suggested to do it at `boards/arm/stm32h7/nucleo-h743zi/src/stm32_appinitialize.c`, I'd suggest to place it at `nuttx/boards/arm/stm32h7/nucleo-h743zi/src/stm32_bringup.c` along with other registrations.

acassis, 2021-07-20:
@saramonteiro and @michallenc the board she is using is the nucleo-h743zi2 not nucleo-h743zi. The file to be modified is the boards/arm/stm32h7/nucleo-h743zi2/src/stm32_bringup.c

eden-desta, 2021-07-20:
Thank you all for taking the time!

So I have attempted to do this as you said above. And these are the steps I took without any success.

the `stm32_bringup.c` file located in `nuttx/boards/arm/stm32h7/nucleo-h743zi2/src`

Already had:
```
#if !defined(CONFIG_ARCH_LEDS) && defined(CONFIG_USERLED_LOWER)
  /* Register the LED driver */

  ret = userled_lower_initialize("/dev/userleds");
  if (ret < 0)
    {
      syslog(LOG_ERR, "ERROR: userled_lower_initialize() failed: %d\n", ret);
    }
#endif
```

So, instead what i did was i edited the `.config` file manually. 
```
CONFIG_USERLED=y
CONFIG_USERLED_LOWER=y
...
CONFIG_ARCH_HAVE_LEDS=y
# CONFIG_ARCH_LEDS is not set
# CONFIG_ARCH_LEDS_CPU_ACTIVITY is not set
```

I had to make some updates to `boards/arm/stm32h7/nucleo-h743zi2/src/stm32_userleds.c` from `GPIO_LED_BLUE` to `GPIO_LED_ORANGE`
In the `stm32_bringup.c` file I had to include `<nuttx/leds/userled.h>`

With all this said and done it build and was able to compile. However, when I run it I still get the same error. 
```
nsh> leds
leds_main: Starting the led_daemon
leds_main: led_daemon started

led_daemon (pid# 3): Running
led_daemon: Opening /dev/userleds
led_daemon: ERROR: Failed to open /dev/userleds: 2
led_daemon: Terminating
```

I ran `ls /dev` and the return value was:
```
nsh> ls /dev
/dev:
 console
 null
 ttyS0
```

acassis, 2021-07-20:
Try to enable in the menuconfig:
Application Configuration  ---> NSH Library  ---> [*]   Have architecture-specific initialization

If it you still facing error (what I don't think will happen), you can enable the LED debug:

Build Setup  --->
    Debug Options  --->
    [*] Enable Debug Features
    [*]   Enable Error Output
    [*]     Enable Warnings Output
    [*]       Enable Informational Debug Output
...
    [*]   Low-level LED Debug Features
    [*]     LED Driver Error Output
    [*]     LED Driver Warnings Output
    [*]     LED Driver Informational Output

eden-desta, 2021-07-21:
@acassis the `Have architecture-specific initializaton` fixed it right up! Thank you!

On the same note is there a place to read more about the menu config options, i tried to look through the nuttx docs and didnt find a lot of detail. Thank you everyone! Welcome to close this out!

acassis, 2021-07-21:
The menuconfig is very dynamic, so any documentation could become outdated very fast. The best way to learn more about the menuconfig is using the "< Help >" button of each feature and the search function: type "/" inside menuconfig and enter the symbol you want to know.
It will show many important things, including: its location inside the menu; the Kconfig inside the source directory tree where it is defined; and from each other symbols the current symbol depends to be available to the users.

You can also use "git grep CONFIG_symbolname" to find where the symbol is used. This way you will learn how this symbol impacts the compilation of the source code.

eden-desta, 2021-07-21:
Oh brilliant! Thank you @acassis

closed 2021-07-21 by eden-desta