apache/nuttx #4189: NuttX md5 command doesn't return same hash value returned by Linux md5sum

acassis, 2021-07-20:
Please see the test:
```
nsh> wapi essid wlan0 ACA2021_2.4GHz 1
nsh> wapi psk wlan0 ******** 3
nsh> ifconfig
wlan0   Link encap:Ethernet HWaddr ac:67:b2:53:95:18 at UP
        inet addr:10.0.0.2 DRaddr:10.0.0.1 Mask:255.255.255.0

nsh> renew wlan0
nsh> ifconfig
wlan0   Link encap:Ethernet HWaddr ac:67:b2:53:95:18 at UP
        inet addr:192.168.0.17 DRaddr:192.168.0.1 Mask:255.255.255.0

nsh> mount                                                                                
  /mnt/esp/wifi type spiffs                                                               
  /proc type procfs                                                                       
nsh> mkrd 256 
nsh> mkfatfs /dev/ram0                                                                    
nsh> mount -t vfat /dev/ram0 /ramdisk/                                                    
nsh> df -h                                                                                
  Filesystem    Size      Used  Available Mounted on                                      
  spiffs        463K     9538B       454K /mnt/esp/wifi                                   
  procfs          0B        0B         0B /proc                                           
  vfat          110K        1K       109K /ramdisk                                        
nsh> cd /ramdisk
nsh> wget http://nginx.org/nginx.png > nginx.png
nsh> ls -l
/ramdisk:
 -rw-rw-rw-    2103 nginx.png
nsh> md5 nginx.png
d2eb2a853ff1dfe9df7ef1fdeec63063
nsh>

alan@dev:~$ md5sum nginx.png 
efabf1246bebb5783d2350492262312d  nginx.png
```

Comparing byte by byte using hexdump on Linux and NuttX, the files are exactly the same.

acassis, 2021-07-20:
An easier way to test:
```
NuttX:

nsh> dd if=/dev/zero of=bytes.bin bs=1 count=8                                            

nsh> md5 bytes.bin                                                                        
1318914cb26155868c8431f54cb7575a                                                          

nsh> hexdump bytes.bin
bytes.bin at 00000000:
0000: 00 00 00 00 00 00 00 00                         ........
nsh>

Linux:

alan@dev:~$ dd if=/dev/zero of=bytes.bin bs=1 count=8
8+0 records in
8+0 records out
8 bytes copied, 0.000684003 s, 11.7 kB/s

alan@dev:~$ md5sum bytes.bin 
7dea362b3fac8e00956a4952a3d4f474  bytes.bin

alan@dev:~$ hexdump -C bytes.bin 
00000000  00 00 00 00 00 00 00 00                           |........|
00000008
```

GUIDINGLI, 2021-07-22:
Please use md5 -f xx.

nsh> help md5
md5 usage:  md5 [-f] <string or filepath>

nuttx will take it as string if you don't add '-f'.

acassis, 2021-07-22:
afff, thank you very much @GUIDINGLI

closed 2021-07-22 by acassis