apache/nuttx #4194: Unnecessary download of ESP Wireless Drivers

TA1DB, 2021-07-21:
I am working with esp32-wrover-kit:wapi configuration. Whenever I do "make menuconfig" and "make", the build system loads the ESP Wireless Drivers again and again irrelevant of their being already downloaded. This is time consuming and loads the network for no reason. I think it will be beneficial if a solution can be found for this issue.

Ouss4, 2021-07-21:
Thanks for reporting this.  I'll be taking a look.

Ouss4, 2021-07-21:
> Whenever I do "make menuconfig" and "make"

Well, there is a `clean_context` in the way, that will delete what was downloaded.  I can fix this but it will break the `context`/`clean_context` symmetry. This is how I had it the very first time, download on `context` and delete on `distclean`, but put some extra effort to not break the symmetry (BTW, apps download packages are also only cleaned on `distclean`).

@xiaoxiang781216 do you have any inputs before I go ahead and do that?

TA1DB, 2021-07-21:
@Ouss4 Thanks for your response. Please note; I use lvgl library as well (for example) , it is downloaded one time only after "make distclean" or in first configuration.

Ouss4, 2021-07-21:
> @Ouss4 Thanks for your response. Please note; I use lvgl library as well (for example) , it is downloaded one time only after "make distclean" or in first configuration.

Yes, this is what I said about "apps packages cleaned on `distclean`", it means they won't be deleted during a `clean_context` which `menuconfig` is doing.

PR #4202 (2021-07-21): arch/: Clean what was made during `context` in distclean.
## Summary
Cleaning during `clean_context` had the issue of remaking everything when `menuconfig` was called.  That's because `menuconfig` has a `clean_context` on its way.

Closes #4194 

## Impact
`make clean_context` won't clean `context` of the arch directory anymore.  We do have this asymmetry in few other places, most notably in apps.

## Testing
Tested with esp32-devkitc:wapi and built few other configs to make sure everything is still working fine.
closed 2021-07-21 by acassis