apache/nuttx #4206: mksyscall need handle 64bit type correctly

xiaoxiang781216, 2021-07-22:
up_syscallx supported by arch can only handle the native integer(uintptr_t) correctly. Since uintptr_t on 32bit arch is 32bit, but  functions listed below need pass or return 64bit value:

1. clock_t clock(void) when CONFIG_SYSTEM_TIME64==y
2. off_t lseek(int fd, off_t offset, int whence) when CONFIG_FS_LARGEFILE==y

mksyscall need modify in this case by:

1. Proxy split 64bit to two 32bit before invoke sys_call
2. Sub merge two 32bit to 64bit before invoke the real function

patacongo, 2021-07-22:
On 64-bit platforms, I believe that both uintptr_t and off_t are 64-bits.  But I am not sure of that.  Need to check.

patacongo, 2021-07-22:
I think this could be easy if:

- In syscall.csv, make all APIs that use off_t depend on !defined(CONFIG_FS_LARGEFILE)
- Add hardcoded implementations that are built only if defined(CONFIG_FS_LARGEFILE)