apache/nuttx #4213: arm_hardfault: PANIC!!

eden-desta, 2021-07-23:
Hello!
I have been recently experimenting with NuttX and found this problem that I cannot quite understand how to fix.
```
arm_hardfault: PANIC!!! Hard fault: 40000000
up_assert: Assertion failed at file:armv7-m/arm_hardfault.c line: 135
up_registerdump: R0: 00000000 08036964 00000000 00000000 38002f90 38002eb0 00000000 380028f0
up_registerdump: R8: 00000000 00000000 00000000 00000000 38002068 380028f0 0802cdcf 08014c1c
up_registerdump: xPSR: 21000000 PRIMASK: 00000000 CONTROL: 00000000
up_registerdump: EXC_RETURN: ffffffe9
up_dumpstate: sp:         38002768
up_dumpstate: stack base: 380014d0
up_dumpstate: stack size: 00001758
up_stackdump: 38002760: 38002768 3800281c 38002eb0 00000000 38002778 080084bd 00001758 380014d0
up_stackdump: 38002780: 38002768 380010c0 38002798 0800856b 00000087 00000000 00000087 08032a90
up_stackdump: 380027a0: 380027a8 08006063 00000087 08032a90 380027b8 08001f05 00000000 00000000
up_stackdump: 380027c0: 3800281c 00000003 00000000 687b0000 08014c1a 3800281c 380027e0 080024a9
up_stackdump: 380027e0: 3800281c 00000003 380027f0 00000003 00000000 08001ea1 38002800 08001e85
up_stackdump: 38002800: 3800281c 00000003 38002810 00000000 380028f0 0800176d 00000010 380028f0
up_stackdump: 38002820: 00000000 38002f90 38002eb0 00000000 380028f0 00000000 00000000 00000000
up_stackdump: 38002840: 00000000 ffffffe9 00000000 00000000 00000000 00000000 00000000 00000000
up_stackdump: 38002860: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
up_stackdump: 38002880: 00000000 00000000 00000000 08036964 00000000 00000000 38002068 0802cdcf
up_stackdump: 380028a0: 08014c1c 21000000 2e5bec55 3fda6f67 00000000 00000000 00000000 00000000
up_stackdump: 380028c0: 00000000 00000000 00000000 00000000 2e5bec55 3fda6f67 ffc00000 41dfffff
up_stackdump: 380028e0: 2e270d87 34dece5c 00000010 080071a3 08036964 00000000 38002f90 38002900
up_stackdump: 38002900: 08036964 24000114 38002910 08013b11 00000014 38003060 38002920 0802cdaf
up_stackdump: 38002920: 08036964 24000114 38002930 0801d01f 08036964 24000114 38002a60 0802cd9d
up_stackdump: 38002940: 38002960 0801b321 240045dc 24004500 38002958 08007d9d 000000c8 24004500
up_stackdump: 38002960: 38002a10 38002f70 24000114 38002eb0 000000d0 00000003 38002f90 00000000
up_stackdump: 38002980: 24001c18 24001c18 240026fc 00000014 38002998 38002f40 38002f90 38002eb0
up_stackdump: 380029a0: 380029c0 08016a81 38002a74 08019209 00000000 00000000 00000000 380029d0
up_stackdump: 380029c0: 08034000 24000114 38002b44 24001378 38002f70 08019209 00000000 38002a60
up_stackdump: 380029e0: 00000001 00000000 38002b4c 00000000 38002a00 08019103 38002a10 00000000
up_stackdump: 38002a00: 08034000 24000114 38002b44 24001378 00000001 0000000a 00000201 00000000
up_stackdump: 38002a20: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
up_stackdump: 38002a40: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
up_stackdump: 38002a60: 080191b9 080191d3 080191eb 08019209 00000000 00000000 24000800 00000000
up_stackdump: 38002a80: 00000000 00000000 38002a98 0801904f 08034a00 00000000 08034000 24000114
up_stackdump: 38002aa0: 38002b44 24001378 38002ab8 08015b75 38002b98 00000000 380014b8 00000001
up_stackdump: 38002ac0: 38002ac8 08001e85 38000230 00000003 dd583223 00000000 38002bb8 0800176d
up_stackdump: 38002ae0: 16a750ef 38002bb8 00000001 00000000 00000000 00000000 38002bb8 00000000
up_stackdump: 38002b00: 00000000 00000000 00000000 ffffffe9 00000000 00000000 00000000 00000000
up_stackdump: 38002b20: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
up_stackdump: 38002b40: 00000000 38002b58 38002df0 38002c30 38002c30 38000230 38002cd0 00000000
up_stackdump: 38002b60: 00000001 00000000 38002b98 01000003 00000000 00000000 08027d47 00000000
up_stackdump: 38002b80: 080191b9 080191d3 080191eb 08019209 00000000 00000000 080191b9 080191d3
up_stackdump: 38002ba0: 080191eb 08019209 00000000 5169fc8d 00000010 38002bb8 38002bc0 00000001
up_stackdump: 38002bc0: 00000000 00000000 00000000 00000000 00000000 08033fd8 38002c48 00000000
up_stackdump: 38002be0: 00000000 00000000 38002bf8 00000000 00000000 0800648d 00000000 380014b8
up_stackdump: 38002c00: 00000001 08015a69 38002c10 08002cff 00000000 380010c0 00000001 00000001
up_stackdump: 38002c20: 00000000 00000000 00000060 80001790 080191b9 080191d3 080191eb 08019209
```

This stack dump occurs when I am trying to initialize a publisher in my application. 
I looked through the `menuconfig` at `RTOS Features --> Stack and heap Information`. I really doubt it is due to stack allocation but I doubled those values for peace of mind. And yet the stack dump is still occurring. 

I am using a `NUCLEO-H743ZI2` board. I have attached my `.config` file below as well.

[Config.txt](https://github.com/apache/incubator-nuttx/files/6868760/Config.txt)

Thank you!! I appreciate your time!

acassis, 2021-07-23:
Hi @eden-desta what is the function at 08014c1c?

Could you please run and post here:
arm-none-eabi-addr2line -e nuttx 0x08014c1c
arm-none-eabi-addr2line -e nuttx 0x0802cdcf

Thank you!

davids5, 2021-07-23:
Have a look at:
https://cwiki.apache.org/confluence/display/NUTTX/Analyzing+Cortex-M+Hardfaults

And https://www.youtube.com/watch?v=KZkAM_PVOi0 Start at ~16 min

eden-desta, 2021-07-23:
@acassis 
` arm-none-eabi-addr2line -e nuttx 0x08014c1c` : `/home/eden/nuttx/nuttx/fs/inode/fs_inoderelease.c:52`
` arm-none-eabi-addr2line -e nuttx 0x0802cdcf` : `/home/eden/nuttx/apps/microros/micro_ros_src/src/rcl/rcl/src/rcl/time.c:455`

@davids5 thank you will look through!

uLipe, 2021-07-23:
It seems a microROS specific problem, have you checked how much memory is being reserved for microROS application in menuconfig @eden-desta ?

Also what is the Nuttx version below the microROS you're using?

eden-desta, 2021-07-26:
@uLipe how do i go about checking how much memory is reserved?

And i installed version 10.1

rushighatkar35, 2021-08-10:
> Hello!
> I have been recently experimenting with NuttX and found this problem that I cannot quite understand how to fix.
>
> ```
> arm_hardfault: PANIC!!! Hard fault: 40000000
> up_assert: Assertion failed [...]

Hi, 

I'm Rushikesh and started to work on Nuttx RTOS with Nucleo-H743ZI2 board.
Could you help me?
Have some doubts and problems while configuring nuttx rtos. Using Ubuntu 20.04

mail id : rushighatkar35@gmail.com

Thanks,