apache/nuttx #4336: Cannot get ip address on bl602evb:wifi

Milili-Tech, 2021-08-13:
I happlen to have a BL602 dev board, very lucky, NuttX also has the support of BL602. So I tried bl602evb:wifi sample to see whether WiFi can work, but unlucky, I cannot get ip address. Just use defconfig of wifi sample, not change anything. The test commonds are:
```
ifup wlan0
wapi mode wlan0 2
wapi psk wlan0 qwerzxcV 3
wapi essid wlan0 Milili-Tech 1
renew wlan0
```
WiFi connection is OK, when excute `renew wlan0`, error message `ERROR: dhcpc_request() failed` occur.

Please give advise if you have tried this sample, thanks.

Virus-V, 2021-08-13:
Try to increase the values of `NETUTILS_DHCPC_RECV_TIMEOUT` and `NETUTILS_DHCPC_RETRIES` in menuconfig.

Milili-Tech, 2021-08-13:
Thanks for your quick feedback, but actually, I captured wireless packets, there are no packets send out from chip when I excute `renew wlan0` command.

acassis, 2021-08-13:
Here it is not connecting:
```
NuttShell (NSH) NuttX-10.1.0
nsh> ?
help usage:  help [-v] [<cmd>]

  .         cp        free      kill      nslookup  rmdir     true      
  [         dirname   help      ls        ps        set       uname     
  ?         df        ifconfig  mkdir     pwd       sleep     umount    
  cat       echo      ifdown    mount     reboot    source    unset     
  cd        false     ifup      mv        rm        test      usleep    

Builtin Apps:
  dhcpd_start  nettest      sh           ping         nsh                                 
  wapi         renew        mm           iperf        ostest                              
  dhcpd_stop   dhcpd        tcpclient    discover                                         
nsh> ifup wlan0                                                                           
ifup wlan0...OK                                                                           
nsh> wapi mode wlan0 2                                                                    
nsh> wapi psk wlan0 MyPassordHere 3                                                            
nsh> wapi essid wlan0 ACA2021_2.4GHz 1                                                    
ioctl(SIOCSIWESSID): Broken pipe                                                          
ERROR: Process command (essid) failed.                                                    
nsh> renew wlan0
```
BTW @Virus-V why LIBCXX is enabled on bl602evb:wifi? It is wasting time to compile your board and it is not used...

Virus-V, 2021-08-13:
@acassis Run `wapi scan wlan0` to see if you can scan your SSID? I suspect that you did not specify the RF parameters when you downloaded the code.
[image]

> BTW @Virus-V why LIBCXX is enabled on bl602evb:wifi? It is wasting time to compile your board and it is not used...
I will check it. If possible, I will remove LIBCXX.

acassis, 2021-08-13:
Hi @Virus-V the scan is working:
```
NuttShell (NSH) NuttX-10.1.0
nsh> ifup wlan0
ifup wlan0...OK
nsh> wapi mode wlan0 2
nsh> wapi scan wlan0
[WIFI] [IND] SCAN Done

wifi_mgmr_scan_complete_callback: scan complete
bssid / frequency / signal level / ssid
08:7e:64:82:d3:20       2412    -39     ACA2021_2.4GHz
60:32:b1:83:1e:da       2442    -62     ol�é telecom.66531_EXT
98:7e:ca:4c:07:bf       2437    -67     VIVOFIBRA-07B1                                    
68:ff:7b:eb:e5:f8       2442    -70     ol�é telecom.66531                                
d8:c6:78:2c:a6:28       2412    -74     se for de paz pode entrar                         
94:ea:ea:b3:0e:4d       2462    -78     VIVOFIBRA-Mariah                                  
12:59:32:ce:54:29       2437    -79                                                       
d8:c6:78:38:36:f0       2462    -79     ATBWF                                             
7c:10:c9:03:10:b1       2437    -98     trojanConvidado                                   
nsh>
```

acassis, 2021-08-13:
I'm not using this graphical interface to flash the board, I use this command:
```
$ blflash flash ./nuttx.bin --port /dev/ttyUSB0
```

Milili-Tech, 2021-08-13:
> @acassis Run `wapi scan wlan0` to see if you can scan your SSID? I suspect that you did not specify the RF parameters when you downloaded the code.
> [image]

I think it's not related to RF parameters, since I also flash wifi examples in BL SDK, it works well.

Virus-V, 2021-08-14:
> I think it's not related to RF parameters, since I also flash wifi example from BL offical SDK, it works well. And same as acassis, I use command tool and not flash any other binary.

The official BL SDK uses the device tree to save the RF parameter information, but NuttX currently does not support the device tree. Therefore, when linking the NuttX of BL602, the linker script will reserve 1K space at the beginning of 0x400 in the generated bin to save the RF parameters obtained from the device tree.

When using the official download tool to download the code, the tool will automatically parse the specified device tree file and find the RF parameter, convert it into a TLV format raw binary data, and write it directly into the 0x400 offset of the bin file. 

As far as I know, third-party tools currently do not support automatic patching of TLV RF parameters in the bin file. 

```
$ hexdump -C -s 0x400 -n 512  nuttx.bin
00000400  42 4c 52 46 50 41 52 41  00 00 00 00 00 00 00 00  |BLRFPARA........|
00000410  00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  |................|
*
00000600
```

acassis, 2021-08-14:
Hi @Virus-V should be nice if your company could submit support for these open-source tools. Normally I avoid to use graphic application because it is not productive for automated testing.

Milili-Tech, 2021-09-02:
@Virus-V Thanks for your info. Tried the GUI tool in a VM, it works now. Hope your guys can provide the command line tool soon.

closed 2021-09-02 by Milili-Tech

xouillet, 2021-10-03:
In the meantime, I've written a small utility to embed the RF parameters in nuttx binary to avoid using BouffaloLabDevCube: https://github.com/xouillet/bl_rfbin