apache/nuttx #4393: Missing semicolon in threads.h

fxschauber, 2021-08-23:
https://github.com/apache/incubator-nuttx/blob/07d528fd8dcbc7b028c636441a9b6f25c23c822c/include/threads.h#L79

xiaoxiang781216, 2021-08-23:
@fxschauber could you provide a patch? Thanks.

acassis, 2021-08-23:
@fxschauber @xiaoxiang781216 I think adding ";" will add two ";" in the code with this CODE, see:
```
CODE int (*au_reset)(FAR struct adc_dev_s *dev);
```

acassis, 2021-08-23:
@fxschauber you are right, looking other examples I can see it is missing the ";"

fxschauber, 2021-08-23:
> @fxschauber could you provide a patch? Thanks.

Sorry for the dumb question but I've never contributed, do I need to fork then do a pull-request? Is there a README of the procedure somewhere?

acassis, 2021-08-23:
@fxschauber it is not a dumb question, in fact it is our fault and we will fix it.

Yes, please create a fork, add your commit to your branch and create a pull request.

Ouss4, 2021-08-23:
> Is there a README of the procedure somewhere?

https://nuttx.apache.org/docs/latest/contributing/making-changes.html

acassis, 2021-08-23:
Thank you @Ouss4 in fact I look there and missed the contribute session, it was at the end... my fault

closed 2021-08-23 by xiaoxiang781216