apache/nuttx #4446: sim:lvgl is not working

acassis, 2021-08-31:
The LVGL example running in the simulator is a good way to test LVGL on NuttX.

Unfortunately after the commit fa0d123f87663c28b432d8b379b8f252bb8411c9 it stopped to work.

Returning to a commit just before that commit works:
$ git reset --hard dbf9c87a428612a2e3374e24081621502eebd822

@xiaoxiang781216 could you please take a look to understand what happened?

I tried to disable CONFIG_STACK_COLORATION but it didn't help.

xiaoxiang781216, 2021-08-31:
Oh, we use lvgl on sim everyday, nobody report the problem. @Donny9 could you take look?

acassis, 2021-08-31:
@xiaoxiang781216 someone told it wasn't working, then I decided to test and confirmed it wasn't working. Then I used "git bisect" to nuttx and apps to identify what caused the issue.

gustavonihei, 2021-08-31:
I've done a quick investigation and the crash is being triggered by the stack protector.
The crash also happens on `sim:nettest`.
I couldn't identify the root cause, but I was wondering if it is correct to have the stack protector enabled on the stack coloration function, since the color pattern will be written outside of the stack boundaries for the `up_stack_color` function.

Disabling the stack protector just for `up_stack_color` prevents the issue.
If this is indeed the correct approach, I can submit the PR: https://github.com/apache/incubator-nuttx/commit/40b136198c67475e83017c4d83c12c73e42d2176 / https://github.com/apache/incubator-nuttx/commit/fe492b9d51d0924743bb269e269047eb9f9febd6

xiaoxiang781216, 2021-09-01:
may cause by SSP, thanks for finding.

PR #4459 (2021-09-01): sim: Inhibit stack protector on stack coloration function
## Summary
This PR intends to inhibit the stack protector on the stack coloration function, which purposely writes outside the `up_stack_color`  function's reserved stack area.
Some `sim` configurations were crashing on the stack protector check.
The exact reason on why the issue does not manifest for every configuration is not clear.

## Impact
Closes #4446 

## Testing
`sim:lvgl` and `sim:nettest` were crashing without the PR changes.
The changes enable them to initialize properly.
closed 2021-09-01 by xiaoxiang781216