apache/nuttx #4481: How to use the "Board Specific drivers" option?

imchangchang, 2021-09-06:
Hi, I  want to define some board-related device drivers.  To do this, I enable the "Board Specific drivers" option under the "Device Drivers" submenu.

I notice that this config will include the **platform/Make.defs**.  The **platform** folder is a link of **BOARD_DRIVERS_DIR** of _tools/Config.mk_.

Related code of **BOARD_DRIVERS_DIR** in _tools/Config.mk_: 
```makefile
BOARD_COMMON_DIR ?= $(wildcard $(BOARD_DIR)$(DELIM)..$(DELIM)common)
BOARD_DRIVERS_DIR ?= $(wildcard $(BOARD_DIR)$(DELIM)..$(DELIM)drivers)
ifeq ($(BOARD_DRIVERS_DIR),)
  BOARD_DRIVERS_DIR = $(TOPDIR)$(DELIM)drivers$(DELIM)dummy
endif
```

From the code, I know that if I want to define some board drivers, I should create a drivers folder in the parent directory of my board. 

For example, if my board folder is **_boards/arm/stm32/fire-stm32v2_** , the drivers folder should be **_boards/arm/stm32/drivers_**.

Is this the right way to use the "Board Specific Drivers"? And why not looking for the driver folder in the board folder?

Ouss4, 2021-09-08:
> Is this the right way to use the "Board Specific Drivers"? 

Yes, the folder needs to be at boards/arm/stm32/drivers, you can take a look at cxd56xx for an example.

> And why not looking for the driver folder in the board folder?

These drivers are supposed to be common to the entire family.

imchangchang, 2021-09-09:
> > Is this the right way to use the "Board Specific Drivers"?
> 
> Yes, the folder needs to be at boards/arm/stm32/drivers, you can take a look at cxd56xx for an example.
> 
> > And why not looking for the driver folder in the board folder?
> 
> These drivers are supposed to be common to the entire family.

Got it, thanks for your reply.

closed 2021-09-09 by imchangchang