apache/nuttx #4510: nucleo-g431kb build error

daniel-p-carvalho, 2021-09-10:
I am trying to build nucleo-g431kb:nsh configuration but I getting the following error
  $arm-none-eabi-ld: Error: unable to disambiguate: -nostartfiles (did you mean --nostartfiles ?)

Probably this is related with commits made on Sep 7 since I can build this configuration up to Sep 6.

Ouss4, 2021-09-10:
This is related to the following:
- https://github.com/apache/incubator-nuttx/pull/3900
- https://github.com/apache/incubator-nuttx/pull/3836
- https://github.com/apache/incubator-nuttx/issues/3826

cc @xiaoxiang781216 @yamt

pkarashchenko, 2022-01-10:
@daniel-p-carvalho please feedback if the issue is still relevant. I tried to rebuild nucleo-g431kb:nsh example and it succeeded.

daniel-p-carvalho, 2022-01-10:
I am out of office... I will check this tomorrow.

pkarashchenko, 2022-05-24:
Is the issue still relevant or it can be closed?

pkarashchenko, 2022-07-04:
@daniel-p-carvalho Is the issue still relevant or it can be closed?

daniel-p-carvalho, 2022-07-04:
Hello Petro,

I am able to build and run nucleo-g431kb:nsh configuration.

Daniel Pereira de Carvalho

closed 2022-07-04 by pkarashchenko