apache/nuttx #4562: quickstart install instructions don't quite work (tar ball top-level dir misnamed)

craigcomstock, 2021-09-16:
Following instructions here: http://nuttx.apache.org/docs/latest/quickstart/install.html
and then http://nuttx.apache.org/docs/latest/quickstart/compiling.html (on OSX, but I don't think that's the issue)

The tarballs have odd names which don't work with the compiling instructions: `apache-incubator-nuttx-apps-d8757a6` and `apache-incubator-nuttx-a5406c6`. If I rename things work fine.

So maybe a change in docs or in tarball generation? Not sure which.

Ouss4, 2021-09-16:
Github is appending the commit SHA to the directories names.  I guess they can just be renamed when extracted.

acassis, 2021-09-16:
@craigcomstock at least on Linux you can use this command:
$ tar zxf apps.tar.gz --one-top-level=apps --strip-components 1

craigcomstock, 2021-09-16:
@acassis yes, I did do that, just wasn't sure which way to go in terms of fixing the docs or tarballs. :+1: I am building happily on osx. Looks like the doc fix is the choice. Thanks!

closed 2021-09-17 by btashton