apache/nuttx #4612: exit() and pthread_exit() limitations

patacongo, 2021-09-26:
The main thread of a task should be able to call pthread_exit(), of course, calling pthread_self() first.  Certainly this behavior is not prohibited by the specifications at OpenGroup.org and there are several posts that specifically claim that this is possible.  See for example:  https://programmer.help/blogs/the-use-of-pthread-exit-in-main-thread.html

In this case, the main thread of the process/task group will terminate, but the process/task group will persist and other threads will continue running.  I believe that calling pthread_exit() from the main thread will currently lead to an assertion or crash due to the different forms of the main and pthread TCBs and the fact that global data is retained in the main thread's stack.

Similarly, a pthread should be able to call exit() and terminate the entire process/  See for example https://www.informit.com/articles/article.aspx?p=2085690&seqNum=5.  In fact, this is often the case when exit() is called from a signal handler in a multi-threaded task because we never know for sure which thread the signal handler will run on.

Again, a crash would be likely due to the same reasoning.

xiaoxiang781216, 2021-09-26:
> The main thread of a task should be able to call pthread_exit(), of course, calling pthread_self() first. Certainly this behavior is not prohibited by the specifications at OpenGroup.org and there [...]

Yes, this is major reason why we can't handle main thread and pthread equally. The best fix could be unify two tcb struct into one and:

1. Move main thread specific data to task_info
2. Move pthread specific data to tls_info

> and the fact that global data is retained in the main thread's stack.
> 

It's fixed by patch: https://github.com/apache/incubator-nuttx/pull/3992

> Similarly, a pthread should be able to call exit() and terminate the entire process/ See for example https://www.informit.com/articles/article.aspx?p=2085690&seqNum=5. In fact, this is often the [...]

patacongo, 2021-09-26:
Wouldn't you also need some kind of reference count on the task data so that it persists until the last thread exits?

xiaoxiang781216, 2021-09-26:
The reference count is already there: group_s(and then task_info_s) get destroyed only after the last thread in the process exit.

patacongo, 2021-09-26:
Should this issue be closed?  Is there no longer an issue?

patacongo, 2021-09-26:
Here is a test case:

```
static void *thread_func(FAR void *parameter)
{
  /* Wait a bit then call exit */

  printf("Child: Started\n");
  sleep(1);

  printf("Child: exit()\n");
  exit(0);
}

int main(int argc, FAR char *argv[])
{
  pthread_t thread;
  int status;

  /* Start the thread instances */

  printf("main: Starting thread\n");
  status = pthread_create(&thread, NULL, thread_func, (pthread_addr_t)NULL);
  if (status != 0)
    {
      printf("ERROR in thread creation: %d\n", status);
    }

  /* And exit the main thread */

  printf("main: pthread_exit()\n");
  pthread_exit(NULL);
}

```
When I run this test, I get:

```
nsh> exit
main: Starting thread
Child: Started
main: pthread_exit()
nsh> Child: exit()

nsh>
```

So I conclude that there is no problem.  Sorry for the false alarm.  I will close this issue now.

closed 2021-09-26 by patacongo

xiaoxiang781216, 2021-09-27:
I could like to add a new issue to unify task and thread tcb into one.