apache/nuttx #4629: Can't get Particle Photon Wi-Fi to work on 3 different versions

quantumhu, 2021-09-30:
I'm having an identical issue to a previously closed issue here: https://github.com/apache/incubator-nuttx/issues/3663

I have the Particle Photon board and am unable to get the WiFi working. I've taken approximately the same steps as the issue opener above did:

```$ git clone https://github.com/apache/incubator-nuttx.git nuttx
$ git clone https://github.com/apache/incubator-nuttx-apps apps
$ cd nuttx
$ ./tools/configure.sh -m photon:wlan
$ make menuconfig
$ # set WAPI config under Application Config > Network Utilities > Network Initialization
$ make
$ sudo dfu-util -d 2b04:d006 -a 0 -s 0x08020000 -D nuttx.bin
```

I tried version 10.1.0, 9.0.0 and 8.2 and all three didn't work. I don't have a serial device to check the boot up logs. Just wanted to check in to see if I missed something obvious.

Thank you!

quantumhu, 2021-09-30:
Update: When I use my phone hotspot, it is able to connect on both versions of NuttX and I'm able to successfully telnet. I assume that my iPhone uses WPA2 Personal, so the only difference between my test WiFi and phone hotspot is that my test WiFi SSID has a dash and numbers.

Is it possible the dash (hypen minus 0x2d) is being encoded improperly or read improperly?

acassis, 2023-03-19:
@quantumhu sorry nobody replied your issue. BTW, now there is a new issue about it here: https://github.com/apache/nuttx/issues/8839 case you still interested.

I will close this duplicated issue.

closed 2023-03-19 by acassis