apache/nuttx #4648: NuttX and FreeBSD / BSD

cederom, 2021-10-08:
Hello world :-)

My first steps with NuttX on FreeBSD. I will ask some questions here for support and make it working on FreeBSD / BSD Unix :-)

I have also asked that question on mailing list but I prefer to work here closer to the source code and better public reach.

My Target hardware is `ESP32-C3` (using `RISC-V` CPU) `ESP32-C3-MINI-1 / ESP32-C3-DevKitM-1` and/or `ESP32-C3-WROOM-02 / ESP32-C3-DevKitC-02`.

First problem is with `kconfig` package:
```
% ./tools/configure.sh esp32c3-devkit:ble
Copy files /home/cd/cederom/work/CeDeROM/lab/embedded/nuttx/nuttx.git/nuttx/tools/../tools/sethost.sh:
line 129: kconfig-tweak: command not found
```

There is no such package kconfig on FreeBSD. But I was working with Zephyr KConfig somehow.

All I have is:
```
pkg search -x kconfig
kf5-kconfig-5.86.0 KF5 widgets for configuration dialogs
kf5-kconfigwidgets-5.86.0 KF5 widgets for configuration dialogs
```
Both are installed.

Any hints welcome :-)
Tomek

cederom, 2021-10-17:
I have managed to build and flash `esp32c3-devkit:ble` on FreeBSD :-) It gets a boot loop problem but the build seem to have worked just the same as on Linux Debian, probably needs some more setup with bootloader :-)

I have tried `./tools/configure.sh -B sim:nsh` but that needs porting `genromfs` to FreeBSD.

`genromfs` was easy peasy :-)

I am using upstream version 0.5.7 from https://github.com/chexum/genromfs

```
0xCFMX4% uname -a
FreeBSD 0xCFMX4 13.0-STABLE FreeBSD 13.0-STABLE #0 stable/13-n247642-39a1ff43ad7: Tue Oct 12 12:31:41 CEST 2021     root@0xCFMX4:/usr/obj/usr/src/amd64.amd64/sys/GENERIC  amd64

0xCFMX4% ls -al nuttx
-rwxr-xr-x  1 XXX  XXX  1516992 Oct 18 01:41 nuttx

0xCFMX4% ./nuttx
login: admin
password:
User Logged-in!

NuttShell (NSH) NuttX-10.1.0
MOTD: username=admin password=Administrator
nsh> uname -a
NuttX 10.1.0 9cb35d79cc-dirty Oct 18 2021 01:40:26 sim sim
nsh> ls
/:
 bin/
 dev/
 etc/
 proc/
 tmp/
nsh>
```

:-)

xiaoxiang781216, 2021-11-22:
@cederom look like all flaw are fixed, can we close this issue?

cederom, 2021-11-29:
Sure! Thank you! I hope it lands into a release soon :-)

closed 2021-11-29 by cederom