apache/nuttx #4845: Network address was changed unexpectedly

SPRESENSE, 2021-11-17:
When connecting to a network with a device such as gs2200m (eth0),
if you try to change the address of the appropriate interface device name (e.g., hoge0) in ifconfig,
the network address of eth0 will be changed.

Is this behavior correct?
[image]

Thank you very much in advance.

acassis, 2021-11-17:
No, it is not a feature, it is a bug! :-D

gustavonihei, 2021-12-21:
This might be related to the usage of USRSOCK component.

`netlib` does not perform any validation on the interface name before requesting the IP change to the network stack.
https://github.com/apache/incubator-nuttx-apps/blob/master/netutils/netlib/netlib_setipv4addr.c#L66

When USRSOCK is enabled, the NuttX netstack is bypassed and the handling of the `SIOCSIFADDR` ioctl command is delegated to the USRSOCK daemon, which in this case is `gs2200md`.
So, in the current state, it is up to the device driver connected via USRSOCK to do the right processing, similar to what `netdev` does:
https://github.com/apache/incubator-nuttx/blob/master/net/netdev/netdev_ioctl.c#L605-L617