apache/nuttx #4919: Error building for Raspberry Pi Pico - Not generating version.h

Aaron-Rumpler, 2021-11-30:
I'm trying to build `raspberrypi-pico:nsh`, however I'm getting the following error:

```
arm-none-eabi-gcc -nostartfiles -nodefaultlibs -T/Users/aaron/Desktop/nuttx/nuttx/boards/arm/rp2040/raspberrypi-pico/scripts/raspberrypi-pico-flash.ld -DPICO_BOARD=\"pico\" -DPICO_BUILD=1 -DPICO_NO_HARDWARE=0 -DPICO_ON_DEVICE=1 -Ichip/boot2 -I/Users/aaron/pico/pico-sdk/src/rp2_common/boot_stage2/asminclude -I/Users/aaron/pico/pico-sdk/src/rp2040/hardware_regs/include -I/Users/aaron/pico/pico-sdk/src/rp2_common/hardware_base/include -I/Users/aaron/pico/pico-sdk/src/common/pico_base/include -I/Users/aaron/pico/pico-sdk/src/boards/include -I/Users/aaron/pico/pico-sdk/src/rp2_common/pico_platform/include -o rp2040_boot_stage2.elf /Users/aaron/pico/pico-sdk/src/rp2_common/boot_stage2/boot2_w25q080.S
In file included from /Users/aaron/pico/pico-sdk/src/rp2_common/pico_platform/include/pico/asm_helper.S:7,
                 from /Users/aaron/pico/pico-sdk/src/rp2_common/boot_stage2/boot2_w25q080.S:29:
/Users/aaron/pico/pico-sdk/src/common/pico_base/include/pico.h:20:10: fatal error: pico/version.h: No such file or directory
   20 | #include "pico/version.h"
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [rp2040_boot_stage2.elf] Error 1
make[1]: *** [.depend] Error 2
make: *** [pass2dep] Error 2
```

`$PICO_SDK_PATH/src/common/pico_base/include/pico/version.h` itself doesn't exist, as it's meant to be generated from `$PICO_SDK_PATH/src/common/pico_base/include/pico/version.h.in`.

I'm running macOS 12.0.1, and have tried with both the `master` branch, as well as the `nuttx-10.2.0` and `nuttx-10.1.0` tags of the NuttX repos.

acassis, 2021-11-30:
Hi @Aaron-Rumpler are you not in the NuttX mailing list? This issue was reported and fixed few days ago: https://www.mail-archive.com/dev@nuttx.apache.org/msg07218.html

If you update your repository the issue should go away.

Please visit https://nuttx.apache.org/community/ and see the instructions to subscribe to the mailing list.

Aaron-Rumpler, 2021-12-01:
Somehow missed that mailing list, even though that thread comes up in the same Google search I tried earlier. Using version 1.1.2 of the Raspberry Pi Pico SDK fixes that particular problem.

closed 2021-12-01 by Aaron-Rumpler