apache/nuttx #4952: Error "Changes not staged for commit". Unable to build.

tito97sp, 2021-12-08:
I can not build nucleo-h743zi changes in #4908 pull request.

Github action replies with this error whenever I push and try to build...

```
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
	modified:   boards/arm/stm32h7/nucleo-h743zi/configs/mcuboot-app/defconfig
	modified:   boards/arm/stm32h7/nucleo-h743zi/configs/mcuboot-loader/defconfig
```

Anybody have had this problem before or knows how to solve it?

xiaoxiang781216, 2021-12-08:
this is because defconfig isn't in the canonical format:
```
====================================================================================
Configuration/Tool: nucleo-h743zi/mcuboot-app,CONFIG_ARMV7M_TOOLCHAIN_GNU_EABIL
------------------------------------------------------------------------------------
  Cleaning...
  Configuring...
  Disabling CONFIG_ARMV7M_TOOLCHAIN_GNU_EABIL
  Enabling CONFIG_ARMV7M_TOOLCHAIN_GNU_EABIL
  Building NuttX...
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
100   159  100   159    0     0   2564      0 --:--:-- --:--:-- --:--:--  2564

100  483k    0  483k    0     0  1991k      0 --:--:-- --:--:-- --:--:-- 1991k
100 1596k    0 1596k    0     0  3662k      0 --:--:-- --:--:-- --:--:-- 5795k
  Normalize nucleo-h743zi/mcuboot-app
8d7
< CONFIG_STM32_FORMAT_MCUBOOT=y
35d33
< CONFIG_MCUBOOT_VERSION="7c890f4b075aed73e4c825ccf875b2fb9ebf2ded"
37,39d34
< CONFIG_MTD=y
< CONFIG_MTD_PARTITION=y
< CONFIG_MTD_PROGMEM=y
80a76
> CONFIG_STM32H7_FLASH_OVERRIDE_I=y
88d83
< CONFIG_STM32H7_FLASH_OVERRIDE_I=y
90a86
> CONFIG_STM32_FORMAT_MCUBOOT=y
Saving the new configuration file
HEAD detached at pull/4908/merge
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
	modified:   boards/arm/stm32h7/nucleo-h743zi/configs/mcuboot-app/defconfig
```
you can run this to normalize it:
./tools/refresh.sh --silent nucleo-h743zi/mcuboot-app

tito97sp, 2021-12-08:
Thanks so much  @xiaoxiang781216. I was becoming mad!

closed 2021-12-08 by tito97sp