apache/nuttx #5020: ramdisk can be created,but mkfatfs /dev/ram0 falied with errno = 6

yanyunyouyou, 2021-12-17:
I used stm32f429i-disco:nsh configuration example ,and just enable the fat filesystem . The ramdisk can be created ,but mkfatfs can not work.  
[image]

anjiahao1, 2021-12-20:
try it:
```
mkrd -m 10 -s 512 1024
mount -t fatfs -o forceformat /dev/ram10 /data
```

pkarashchenko, 2022-01-08:
@yanyunyouyou is the issue still relevant or you are satisfied with the answer?

yanyunyouyou, 2022-01-11:
[image]
It took me a long time to solve this problem. The main reason is that I use WSL and vscode. Some incorrect configurations lead to incorrect header files. I changed to Ubuntu 18.04, recompiled, and the command can be executed. Thank anjiahao1 for the advice, but the answer is wrong. Please see my screenshot.

yanyunyouyou, 2022-01-11:
> @yanyunyouyou is the issue still relevant or you are satisfied with the answer?
     
[image]
It took me a long time to solve this problem. The main reason is that I use WSL and vscode. Some incorrect configurations lead to incorrect header files. I changed to Ubuntu 18.04, recompiled, and the command can be executed. Thank anjiahao1 for the advice, but the answer is wrong. Please see my screenshot.

pkarashchenko, 2022-01-11:
Yes. `mount` does not know what is `-t fatfs` and `vfat` does not have `-o forceformat` option (it is from `littlefs`).
So if I understand correctly the issue is resolved and can be closed?

yanyunyouyou, 2022-01-12:
Yes，it can be closed now。Thank you。

closed 2022-01-12 by pkarashchenko

chenzhihong007, 2023-05-05:
In my case, nsh can't found mkfatfs command. How can I configuration to enable mkfatfs? Thank you!
[image]

TimJTi, 2023-05-05:
From menuconfig, type a forward slash and it will open a search function. Type MKFATFS and it will show you that FSUTILS_MKFATFS is what you need, and it is dependent on FS_FAT etc. It even tells you if those dependencies are enabled.

 

You can type the number of the relevant search result (1, in this case) and it will take you directly to the menuconfig entry, assuming the dependencies allow it to be visible.

 

HTH!

 

 <https://user-images.githubusercontent.com/109498998/236416066-e17894a3-7346-4191-b94a-134514378e1a.png>

yanyunyouyou, 2023-05-07:
nuttx玩起了太累了，加油吧

 
I got it, thank you very much!