apache/nuttx #5068: TCP/UDP source port selection should be random

normanr, 2021-12-23:
Currently the TCP/UDP port base is selected with:
```
g_last_tcp_port = clock_systime_ticks() % 32000;
if (g_last_tcp_port < 4096)
  {
    g_last_tcp_port += 4096;
  }
```
and "the next available port" is just incremented from there (with wrap around). This results in devices using the same ports when they're started up (which causes weird results because servers think connections are already open when they're not).

It would be better to use a random number on every selection (instead of plain increment starting from a time based number) as many platforms (even tiny embedded ones) have true random number generators these days and can seed the random pool in `board_init_rngseed()`

adiroy05, 2022-01-19:
I will fix it assigne to me

xiaoxiang781216, 2022-05-15:
Fixed by https://github.com/apache/incubator-nuttx/pull/6154

closed 2022-05-15 by xiaoxiang781216