apache/nuttx #5142: FlexCAN can_input() should not be called directly from interrupt handler

raiden00pl, 2022-01-03:
I noticed that FlexCAN implementation calls can_input() directly from the interrupt handler which is probably wrong. 
This will raise the assertion in nxsem_trywait.c:75 when CONFIG_DEBUG_ASSERTIONS=y 

The call stack looks like this: xxx_flexcan_interrupt -> xxx_receive -> can_input -> can_callback -> net_trylock -> nxsem_trywait.

I don't have any NXP boards to help solve the problem, so I just open a new issue.

raiden00pl, 2022-01-03:
FYI @PetervdPerk-NXP @michallenc

PetervdPerk-NXP, 2022-01-03:
Interesting the comment above the function suggest it wouldn't block and didn't specify it's not allowed to be executed while in an interrupt. https://github.com/apache/incubator-nuttx/blob/9a53601ba993d43684328dcae8750acb00a34cc7/sched/semaphore/sem_trywait.c#L45-L48

The reason why it's called under an interrupt is because Unfortunely using a workqueue it's too slow to react and would cause an CAN RX mailbox overflow. That's why we try to get rid of asap while in the interrupt but only if we can access the lock hence the net_trylock().

A better solution would to figure why scheduling a workqueue with an immediate deadline takes quite to long to be scheduled but I never came to that.

raiden00pl, 2023-08-04:
I'm reopening this issue as it is not fully resolved yet

closed 2023-08-14 by xiaoxiang781216

raiden00pl, 2023-08-16:
I don't have any NXP board so I can't confirm if the problem has been fixed (it has to be tested on HW). Let someone who works with NXP confirm that the problem is fixed.