apache/nuttx #5184: Replace all place which use semaphore as lock with mutex wrapper

xiaoxiang781216, 2022-01-06:
See the discussion: https://github.com/apache/incubator-nuttx/pull/5070 and https://www.mail-archive.com/dev@nuttx.apache.org/msg07312.html

pkarashchenko, 2022-01-08:
@xiaoxiang781216 will we improve mutex wrapper to check for holder before releasing a mutex or a separate ticket should be created for that? Currently mutex can be released from a task that is not holding it.

xiaoxiang781216, 2022-01-09:
it can be added in the separate PR. Maybe need to enabled in the debug mode to save the space.

pkarashchenko, 2022-01-22:
I think that probably a good way will be to move priority inheritance from sem to mutex and clean-up priority inheritance APIs for semaphores after we switch to mutexes in code

xiaoxiang781216, 2022-01-22:
Yes, this is the safe path to fix the priority inheritance issue.

closed 2022-10-22 by xiaoxiang781216