apache/nuttx #5255: make: *** No rule to make target 'byte', needed by 'differ'.  Stop.

pkarashchenko, 2022-01-17:
Sometimes when running `make` few times in a row compilation fails with `make: *** No rule to make target 'byte', needed by 'differ'.  Stop.`

Ouss4, 2022-01-18:
Can you grab the output of `make V=1` when that happens?

xiaoxiang781216, 2022-01-18:
I hit too, but pass in the next invocation.

raiden00pl, 2022-01-22:
Just had a luck to capture this error with `make V=1`. Unfortunately, there is nothing useful to see. Absolutely nothing :)

```
[raiden00:~/git/RTOS/nuttx/nuttx]$ make V=1                                                                                                                                                                                                                                         
make: *** No rule to make target 'byte', needed by 'differ'.  Stop.
```

fjpanag, 2022-01-24:
I am getting the same error, but only on my Fedora **35** machine.  
On Fedora **34** this issue does not appear.  
(Both systems are fully updated.)

I don't know if this is of any help.

AlanRosenthal, 2022-01-24:
Please run `V=1 make --context`. it might be a lot of output

closed 2022-01-26 by pkarashchenko