apache/nuttx #5324: tools/Unix.mk:49: *** missing separator.  Stop.

olegk0, 2022-01-24:
when i try run make it fails with error 

**tools/Unix.mk:49: *** missing separator.  Stop.**

it probably depends on build tools versions

I solved my problem in the following way 

```
--- a/Unix.mk	2022-01-24 12:57:33.708192426 +0300
+++ b/Unix.mk	2022-01-24 12:57:17.859721267 +0300
@@ -46,8 +46,9 @@
 # Only update .version if the contents of version.tmp actually changes
 # Note: this is executed before any rule is run
 
-$(shell tools/version.sh .version.tmp)
+$(shell `tools/version.sh .version.tmp`)

```

AlanRosenthal, 2022-01-24:
@olegk0 what version of make are you using? What is your system setup?

olegk0, 2022-01-24:
> @olegk0 what version of make are you using? What is your system setup?

GNU Make 4.3
Debian GNU/Linux 11 (bullseye)

AlanRosenthal, 2022-01-24:
According to the doc: https://www.gnu.org/software/make/manual/html_node/Shell-Function.html
the `shell` function does the same thing as the backtick.

what command are you running?

olegk0, 2022-01-24:
> what command are you running?

make

make clean

AlanRosenthal, 2022-01-24:
that script needs a param passed to it:
```
➜ tools/version.sh
Missing path to the output file
USAGE: tools/version.sh [-d|-h] [-b <build>] [-v <major.minor.patch>] <outfile-path>
Try 'tools/version.sh -h' for more information
```

can you try running it in a super clean env? git clean -xffd

olegk0, 2022-01-25:
> that script needs a param passed to it:

I gave an unsuccessful example, I meant any valid command without a parameter 

> can you try running it in a super clean env? git clean -xffd

```
$ git clean -xffd
...
Removing .config
Removing .version.tmp
Removing Make.defs
Removing defconfig
Removing tools/incdir

$ tools/configure.sh -l qemu-intel64:nsh
  Copy files
  Select CONFIG_HOST_LINUX=y
  Refreshing...
tools/Unix.mk:49: *** missing separator.  Stop.
ERROR: failed to refresh

$ make
tools/Unix.mk:49: *** missing separator.  Stop.

```

anchao, 2022-02-10:
> $ make
> tools/Unix.mk:49: *** missing separator.  Stop.
> ```

I can reproduce this issue if the git repo without tag:
```
$ git --version
git version 2.25.1

$ git tag | xargs git tag -d
Deleted tag 'nuttx-1.0' (was 0bed0eea9d)
Deleted tag 'nuttx-1.1' (was b259aced01)
Deleted tag 'nuttx-1.2' (was a530c85e88)
Deleted tag 'nuttx-10.0.0' (was c8d523c5f4)
....

$ tools/configure.sh -l qemu-intel64:nsh
  Copy files
  Select CONFIG_HOST_LINUX=y
  Refreshing...
tools/Unix.mk:49: *** missing separator.  Stop.
ERROR: failed to refresh
```

Please check git tag is exist in your repository: 

```
$ git fetch origin 
From github.com:apache/incubator-nuttx
 * [new tag]               nuttx-1.0        -> nuttx-1.0
 * [new tag]               nuttx-1.1        -> nuttx-1.1
 * [new tag]               nuttx-1.2        -> nuttx-1.2
....

$ tools/configure.sh -l qemu-intel64:nsh
  Copy files
  Select CONFIG_HOST_LINUX=y
  Refreshing...
make[1]: Entering directory '/home/archer/code/nuttx/server/incubator-nuttx'
make[2]: Entering directory '/home/archer/code/nuttx/server/incubator-nuttx/boards'
make[2]: Nothing to be done for 'clean_context'.
....
```

olegk0, 2022-02-11:
> Please check git tag is exist in your repository:

Yes, indeed I did not have tags.

I tried to take the full repository, the problem is gone.

Sorry for disturbing.

closed 2022-02-11 by olegk0