apache/nuttx #5377: esp32c3 uart1 not working

fansongsen, 2022-01-30:
https://github.com/apache/incubator-nuttx/blob/e0b62bf677340dbaa472ad8c535d0483cbf65cf7/arch/risc-v/src/esp32c3/esp32c3_serial.c#L309

I think here should open the clock bit SYSTEM_UART1_CLK_EN of uart1,because it was disabled by esp32c3_perip_clk_init

acassis, 2022-02-09:
@fansongsen thanks for reporting it, sorry my delay to see this issue your posted here. I'll do some tests and let you know.

acassis, 2022-02-10:
@fansongsen please let me know if it fixed the issue for you, thank you!

closed 2022-02-11 by xiaoxiang781216

XuNeo, 2023-05-14:
@acassis Hi Alan, 

Could you help to check if this issue still exists? I tried master and nuttx-10.3.0-RC0, UART1 seems not working on both branches. Not sure if it's related to my setup or a common issue.

To reproduce,

Based on configuration, `tools/configure.sh boards/risc-v/esp32c3/esp32c3-devkit/configs/usbconsole`, enable `+CONFIG_ESP32C3_UART1=y`,

On device, It will stuck when writing data to ttyS0. Backtrace shows it's waiting in function `uart_tcdrain`.

```
nsh> ls /dev
/dev:
 console
 null
 ttyACM0
 ttyS0
 zero
nsh> echo 123 > /dev/ttyS0 ## stuck here
```