apache/nuttx #5380: Support for Heltec Wifi Kit 32  (ESP32)

snarkipus, 2022-01-30:
Working ESP-IDF Installation - [Successful ](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/get-started/)

Build the esp32-devkitc:nsh project using two different approaches ([here ](https://blog.espressif.com/getting-started-with-esp32-and-nuttx-fd3e1a3d182c)and [here](https://embetronicx.com/tutorials/rtos/nuttx/setup-nuttx-rtos-on-esp32-dev-board/)).

Seems to boot but the serial output is garbled where I would expect to see the shell. The only difference of note would be the install board XTAL is 26MHz vs. the default of 40MHz. The bootloader was configured with the 26MHz option - perhaps nuttx doesn't support this?

`ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6752
load:0x40078000,len:14796
load:0x40080400,len:3792
entry 0x40080694
I (31) boot: ESP-IDF v5.0-dev-1426-ge899edd793 2nd stage bootloader
I (31) boot: compile time 12:28:12
I (31) boot: chip revision: 1
I (35) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (42) boot.esp32: SPI Speed      : 40MHz
I (47) boot.esp32: SPI Mode       : DIO
I (51) boot.esp32: SPI Flash Size : 8MB
I (56) boot: Enabling RNG early entropy source...
I (61) boot: Partition Table:
I (65) boot: ## Label            Usage          Type ST Offset   Length
I (72) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (80) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (87) boot:  2 factory          factory app      00 00 00010000 00100000
I (95) boot: End of partition table
I (99) boot_comm: chip revision: 1, min. application chip revision: 0
I (106) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=01a98h (  6808) map
I (117) esp_image: segment 1: paddr=00011ac0 vaddr=3ffb1310 size=000cch (   204) load
I (123) esp_image: segment 2: paddr=00011b94 vaddr=40080000 size=0174ch (  5964) load
I (134) esp_image: segment 3: paddr=000132e8 vaddr=00000000 size=0cd30h ( 52528)
I (159) esp_image: segment 4: paddr=00020020 vaddr=400d0020 size=0ef68h ( 61288) map
I (182) boot: Loaded app from partition at offset 0x10000
I (182) boot: Disabling RNG early entropy source...
�rl|���rn�p�n
              bll`b�l��bp��b|
                                l`|rn
                                       �nn�`

Ouss4, 2022-01-31:
Can you please configure the onboard crystal to be 26 MHz (make menuconfig -> System Type -> On-board Crystal Frequency) and then apply this patch:

```diff
diff --git a/arch/xtensa/src/esp32/esp32_clockconfig.c b/arch/xtensa/src/esp32/esp32_clockconfig.c
index 27a0a0a9c6..89f30ce901 100644
--- a/arch/xtensa/src/esp32/esp32_clockconfig.c
+++ b/arch/xtensa/src/esp32/esp32_clockconfig.c
@@ -25,6 +25,8 @@
 #include <nuttx/config.h>
 #include <stdint.h>

+#include <arch/board/board.h>
+
 #include "xtensa.h"
 #include "xtensa_attr.h"
 #include "hardware/esp32_dport.h"
@@ -186,7 +188,7 @@ void esp32_clockconfig(void)
   uint32_t freq_mhz = CONFIG_ESP32_DEFAULT_CPU_FREQ_MHZ;
   uint32_t old_freq_mhz;
   uint32_t source_freq_mhz;
-  enum esp32_rtc_xtal_freq_e xtal_freq = RTC_XTAL_FREQ_40M;
+  enum esp32_rtc_xtal_freq_e xtal_freq = BOARD_XTAL_FREQUENCY / MHZ;

   old_freq_mhz = esp_rtc_clk_get_cpu_freq();
   if (old_freq_mhz == freq_mhz)
```

snarkipus, 2022-01-31:
That worked. Thank you.

closed 2022-01-31 by snarkipus