apache/nuttx #5412: Calling mq_send() from an interrupt triggers assertion

szymonnnm, 2022-02-04:
Hello everyone,

I have a problem with the message queue driver. I am currently porting our project of the current version of NuttX. So up until now, our project was using NuttX 8.2 and I am working on making it functional with NuttX 10.2. 

Anyway, in one part of our project, we want to put some data on a message queue from an interrupt service routine. 
```c
static int stm32_ubx_tp_irq_handler(int irq, FAR void *context, FAR void *arg)
{
  struct data_to_send dummy;
  (....)
  mq_send(mq, (char*)&dummy, sizeof(struct data_to_send), 0);
  return 0;
}
```

And it worked flawlessly on NuttX 8.2, but after porting the project to NuttX 10.2 this operation triggers assertion. 
[To be specific, this is the assertion that is being triggered](https://github.com/apache/incubator-nuttx/blob/master/sched/semaphore/sem_wait.c#L78):
```c
int nxsem_wait(FAR sem_t *sem)
{
  FAR struct tcb_s *rtcb = this_task();
  irqstate_t flags;
  int ret = -EINVAL;

  /* This API should not be called from interrupt handlers */

  DEBUGASSERT(sem != NULL && up_interrupt_context() == false);
```

So, I have looked into the history of changes and it turns out that between 8.2 and 10.2 the function `nxmq_send()` has been modified.  In the older version, it was only executing the logic related to putting an item in the message queue, but now it has to obtain the file struct instance first, and obtaining it using regular mechanisms fails because of the fact that we are doing it from an ISR.
The current call order is like this:
`mq_send -> nxmq_send -> fs_getfilep -> _files_semtake -> nxsem_wait_uninterruptible -> nxsem_wait -> ASSERTION TRIGGERED`

So my question is, what is the correct way on NuttX 10.2 to send an item using a message queue from ISR? [The documentation doesn't say anything about it not being possible](https://cwiki.apache.org/confluence/display/NUTTX/User+Guide#Message_Queue). I was also not able to find an alternative API (some RTOSes I have worked with in the past had two separate functions for that, like for example `msgq_send()` and `msgq_send_from_isr()`)

xiaoxiang781216, 2022-02-04:
> So, I have looked into the history of changes and it turns out that between 8.2 and 10.2 the function `nxmq_send()` has been modified. In the older version, it was only executing the logic related [...]

The change of call chain is due to: https://github.com/apache/incubator-nuttx/pull/2631.
The final goal is to support poll on mqueue: https://github.com/apache/incubator-nuttx/pull/2957.

> So my question is, what is the correct way on NuttX 10.2 to send an item using a message queue from ISR? [The documentation doesn't say anything about it not being [...]

It isn't enough to just add msgq_send_from_isr because mqueue handler belong to a particular task, and it's impossible to use mqueue handler directly in the interrupt context.

Instead, you need call file_mq_xxx in kernel side, but mq_xxx in user space.
Please follow audio_null.c change in #2631 to adapter your driver code.

patacongo, 2022-02-04:
Yes, NuttX always supporting sending messages from interrupt handlers.  
Sorry to hear that has been broken.

The original IPC philosophy was that the providing side of all IPCs 
should work from interrupt handlers:  messages, semaphores, signals, etc.

szymonnnm, 2022-02-07:
@xiaoxiang781216  @patacongo 
Thanks a lot for your comments. I modified my code a little bit to change the function that is called from the ISR to `file_mq_send()` and in fact - it works now and doesn't trigger any assertions.

closed 2022-02-07 by szymonnnm