apache/nuttx #5626: Redirects doesn't work with programs like pwm

acassis, 2022-02-26:
I was debugging pwm and to get rid of mixed messages from pwm application and driver debug messages I decided to use "> /dev/null" but for some reason it just returns without executing:

```
nsh> pwm -p /dev/pwm1 -c 3 > /dev/null
nsh>
```
Seems some issue on pwm side because commands like this work fine:

```
nsh> sleep 5 > /dev/null
nsh>
```

pkarashchenko, 2022-03-01:
@acassis is `CONFIG_FILE_STREAM` set in your configuration?

acassis, 2022-03-03:
Hi Petro, CONFIG_FILE_STREAM as set correctly, seem to problem is fixed on current mainline, I was using an old version:
NuttX:
19beb307dd5023c2bf414dc54c3141bfc95c7251

Apps:
c222043ed1ede9ef8e9e93ffed14ee9c41c4a2b1

It can be tested using this config: https://pastebin.com/raw/duUsX6hX

running: pwn -p /dev/pwm0 the orange LED will turn on, but pwm -p /dev/pwm0 > /dev/null will return without executing

closed 2022-03-03 by acassis