apache/nuttx #5832: STM3240-G Build is Broken

patacongo, 2022-03-23:
```
CC:  board/stm32_buttons.c
In file included from board/stm32_buttons.c:34:0:
board/stm3240g-eval.h:106:26: error: 'GPIO_INPUT' undeclared here (not in a function); did you mean 'MAX_INPUT'?
 #define GPIO_BTN_WAKEUP (GPIO_INPUT|GPIO_FLOAT|GPIO_EXTI|GPIO_PORTA|GPIO_PIN0)
                          ^
board/stm32_buttons.c:56:3: note: in expansion of macro 'GPIO_BTN_WAKEUP'
   GPIO_BTN_WAKEUP, GPIO_BTN_TAMPER, GPIO_BTN_USER
   ^~~~~~~~~~~~~~~
board/stm3240g-eval.h:106:37: error: 'GPIO_FLOAT' undeclared here (not in a function); did you mean 'GPIO_INPUT'?
 #define GPIO_BTN_WAKEUP (GPIO_INPUT|GPIO_FLOAT|GPIO_EXTI|GPIO_PORTA|GPIO_PIN0)
                                     ^
board/stm32_buttons.c:56:3: note: in expansion of macro 'GPIO_BTN_WAKEUP'
   GPIO_BTN_WAKEUP, GPIO_BTN_TAMPER, GPIO_BTN_USER
   ^~~~~~~~~~~~~~~
board/stm3240g-eval.h:106:48: error: 'GPIO_EXTI' undeclared here (not in a function); did you mean 'GPIO_LED1'?
 #define GPIO_BTN_WAKEUP (GPIO_INPUT|GPIO_FLOAT|GPIO_EXTI|GPIO_PORTA|GPIO_PIN0)
                                                ^
board/stm32_buttons.c:56:3: note: in expansion of macro 'GPIO_BTN_WAKEUP'
   GPIO_BTN_WAKEUP, GPIO_BTN_TAMPER, GPIO_BTN_USER
   ^~~~~~~~~~~~~~~
board/stm3240g-eval.h:106:58: error: 'GPIO_PORTA' undeclared here (not in a function); did you mean 'GPIO_EXTI'?
 #define GPIO_BTN_WAKEUP (GPIO_INPUT|GPIO_FLOAT|GPIO_EXTI|GPIO_PORTA|GPIO_PIN0)
                                                          ^
board/stm32_buttons.c:56:3: note: in expansion of macro 'GPIO_BTN_WAKEUP'
   GPIO_BTN_WAKEUP, GPIO_BTN_TAMPER, GPIO_BTN_USER
   ^~~~~~~~~~~~~~~
board/stm3240g-eval.h:106:69: error: 'GPIO_PIN0' undeclared here (not in a function); did you mean 'GPIO_PORTA'?
 #define GPIO_BTN_WAKEUP (GPIO_INPUT|GPIO_FLOAT|GPIO_EXTI|GPIO_PORTA|GPIO_PIN0)
                                                                     ^
board/stm32_buttons.c:56:3: note: in expansion of macro 'GPIO_BTN_WAKEUP'
   GPIO_BTN_WAKEUP, GPIO_BTN_TAMPER, GPIO_BTN_USER
   ^~~~~~~~~~~~~~~
board/stm3240g-eval.h:107:58: error: 'GPIO_PORTC' undeclared here (not in a function); did you mean 'GPIO_PORTA'?
 #define GPIO_BTN_TAMPER (GPIO_INPUT|GPIO_FLOAT|GPIO_EXTI|GPIO_PORTC|GPIO_PIN13)
                                                          ^
board/stm32_buttons.c:56:20: note: in expansion of macro 'GPIO_BTN_TAMPER'
   GPIO_BTN_WAKEUP, GPIO_BTN_TAMPER, GPIO_BTN_USER
                    ^~~~~~~~~~~~~~~
board/stm3240g-eval.h:107:69: error: 'GPIO_PIN13' undeclared here (not in a function); did you mean 'GPIO_PIN0'?
 #define GPIO_BTN_TAMPER (GPIO_INPUT|GPIO_FLOAT|GPIO_EXTI|GPIO_PORTC|GPIO_PIN13)
                                                                     ^
board/stm32_buttons.c:56:20: note: in expansion of macro 'GPIO_BTN_TAMPER'
   GPIO_BTN_WAKEUP, GPIO_BTN_TAMPER, GPIO_BTN_USER
                    ^~~~~~~~~~~~~~~
board/stm3240g-eval.h:108:58: error: 'GPIO_PORTG' undeclared here (not in a function); did you mean 'GPIO_PORTC'?
 #define GPIO_BTN_USER   (GPIO_INPUT|GPIO_FLOAT|GPIO_EXTI|GPIO_PORTG|GPIO_PIN15)
                                                          ^
board/stm32_buttons.c:56:37: note: in expansion of macro 'GPIO_BTN_USER'
   GPIO_BTN_WAKEUP, GPIO_BTN_TAMPER, GPIO_BTN_USER
                                     ^~~~~~~~~~~~~
board/stm3240g-eval.h:108:69: error: 'GPIO_PIN15' undeclared here (not in a function); did you mean 'GPIO_PIN13'?
 #define GPIO_BTN_USER   (GPIO_INPUT|GPIO_FLOAT|GPIO_EXTI|GPIO_PORTG|GPIO_PIN15)
                                                                     ^
board/stm32_buttons.c:56:37: note: in expansion of macro 'GPIO_BTN_USER'
   GPIO_BTN_WAKEUP, GPIO_BTN_TAMPER, GPIO_BTN_USER
                                     ^~~~~~~~~~~~~
board/stm32_buttons.c: In function 'board_button_initialize':
board/stm32_buttons.c:84:7: warning: implicit declaration of function 'stm32_configgpio'; did you mean 'stm32_bringup'? [-Wimplicit-function-declaration]
       stm32_configgpio(g_buttons[i]);
       ^~~~~~~~~~~~~~~~
       stm32_bringup
board/stm32_buttons.c: In function 'board_buttons':
board/stm32_buttons.c:107:23: warning: implicit declaration of function 'stm32_gpioread'; did you mean 'stm32_bringup'? [-Wimplicit-function-declaration]
       bool released = stm32_gpioread(g_buttons[i]);
                       ^~~~~~~~~~~~~~
                       stm32_bringup

```

patacongo, 2022-03-23:
Another failure:

```
LD: nuttx
/cygdrive/d/Spuda/Documents/projects/nuttx/master/nuttx/staging/libarch.a(stm32_sdio.o): In function `stm32_lock':
stm32_sdio.c:(.text+0x3a6): undefined reference to `stm32_muxbus_sdio_lock'
```

davids5, 2022-03-23:
@patacongo - what is the configure line used?

patacongo, 2022-03-24:
> @patacongo - what is the configure line used?

It was a custom configuration based on stm3240g-evel:nsh

patacongo, 2022-03-24:
Also, after getting a clean build by disabling buttons and SDIO mux, the resulting binary would not run.  It hardfaulted very early.