apache/nuttx #5916: libc++ on sim is broken

gustavonihei, 2022-03-30:
Testing `libc++` via a slight modified `sim:cxxtest` results in a segmentation fault.
After a post-mortem analysis, it crashes due to a NULL-pointer dereference on the following line:

https://github.com/apache/incubator-nuttx/blob/379639a371697ba0bedbb371e6e437864e0ff71d/sched/semaphore/sem_wait.c#L101

The crash is easily reproducible using the following defconfig:
```
CONFIG_ARCH="sim"
CONFIG_ARCH_BOARD="sim"
CONFIG_ARCH_BOARD_SIM=y
CONFIG_ARCH_CHIP="sim"
CONFIG_ARCH_FLOAT_H=y
CONFIG_ARCH_SIM=y
CONFIG_BOARD_LOOPSPERMSEC=100
CONFIG_DEBUG_FULLOPT=y
CONFIG_DEBUG_SYMBOLS=y
CONFIG_DEFAULT_TASK_STACKSIZE=8192
CONFIG_HAVE_CXX=y
CONFIG_IDLETHREAD_STACKSIZE=8192
CONFIG_INIT_ENTRYPOINT="cxxtest_main"
CONFIG_LIBCXX=y
CONFIG_LIBCXXABI=y
CONFIG_PTHREAD_MUTEX_TYPES=y
CONFIG_RAM_START=0x00000000
CONFIG_SCHED_ATEXIT=y
CONFIG_SCHED_ONEXIT=y
CONFIG_START_DAY=27
CONFIG_START_MONTH=2
CONFIG_START_YEAR=2007
CONFIG_TESTING_CXXTEST=y
```

rtucker85, 2022-03-31:
I also get this error.
I have noticed that if you deselect CONFIG_HAVE_CXXINITIALIZE it doesn't crash.
This may help with hunting down the root cause of this problem.

gustavonihei, 2022-03-31:
> I have noticed that if you deselect CONFIG_HAVE_CXXINITIALIZE it doesn't crash.

The crash on the described scenario still happens even when deselecting `HAVE_CXXINITIALIZE`, because the constructors are being initialized during `glibc` startup process, i.e. before NuttX simulator startup code.

The build script already does some trickery to prevent this premature initialization, but since **glibc 2.34** it is no longer effective due to a refactor on the **glibc** startup process.

PR https://github.com/apache/incubator-nuttx/pull/5923 intends to once again restore the same trick.