apache/nuttx #5942: ESP32-C3 Broken

Ouss4, 2022-04-01:
Looks like recent RISC-V changes broke ESP32-C3 (at least), getting this error in a loop:
```
up_assert: Assertion failed at file:common/riscv_exception.c line: 88 task: Idle Task
riscv_exception: EXCEPTION: Store/AMO access fault. MCAUSE: 00000007
riscv_exception: PANIC!!! Exception = 00000007
```

Not working on this right now, anyone has an idea before taking a look?

gustavonihei, 2022-04-01:
I've just performed a `git bisect`, testing each commit with `esp32c3-devkit:nsh`, and it points to the following commit:

https://github.com/apache/incubator-nuttx/commit/32fe25278aec08ce71c0bcaff73697a591e8e4ad

xiaoxiang781216, 2022-04-01:
@no1wudi could you take a look?

closed 2022-04-01 by acassis