apache/nuttx #5956: LVGL on SIM is crashing

acassis, 2022-04-01:
Some recent modification in the mainline is crashing LVGL demo:
```
alan@dev:/tmp/nuttx$ ./tools/configure.sh sim:lvgl

alan@dev:/tmp/nuttx$ ./nuttx 
Segmentation fault (core dumped)

$ uname -a
Linux dev 5.13.0-39-generic #44-Ubuntu SMP Thu Mar 24 15:35:05 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
```

gustavonihei, 2022-04-01:
After a quick inspection, I noticed the issue does not happen if `MEMSET_OPTSPEED` is selected.

acassis, 2022-04-02:
Doing bisect I discovered that the issue was caused by commit ee931c137fb795630b4c68f040861583d52a3efd
Adding -fno-builtin back to CFLAGS fixed the issue:
```
diff --git a/boards/sim/sim/sim/scripts/Make.defs b/boards/sim/sim/sim/scripts/Make.defs
index fb6bad3a97..abaa5bb2bd 100644
--- a/boards/sim/sim/sim/scripts/Make.defs
+++ b/boards/sim/sim/sim/scripts/Make.defs
@@ -67,8 +67,8 @@ endif
 
 # Add -fno-common because macOS "ld -r" doesn't seem to pick objects
 # for common symbols.
-ARCHCFLAGS = -fno-common -ffunction-sections -fdata-sections
-ARCHCXXFLAGS = -fno-common -ffunction-sections -fdata-sections -nostdinc++
+ARCHCFLAGS = -fno-common -fno-builtin -ffunction-sections -fdata-sections
+ARCHCXXFLAGS = -fno-common -fno-builtin -ffunction-sections -fdata-sections -nostdinc++
 
 ifeq ($(CONFIG_CXX_EXCEPTION),)
   ARCHCXXFLAGS += -fno-exceptions -fcheck-new
```
@xiaoxiang781216 could you please verify?

xiaoxiang781216, 2022-04-02:
Ok, I will take a look tomorrow.

xiaoxiang781216, 2022-04-03:
@acassis fix here: #5971. Please verify in your environment again.

closed 2022-04-05 by pkarashchenko