apache/nuttx #5969: NuttX configure.sh crashing

dosadi, 2022-04-02:
Hi,

I have a brand new NuttX pull, and when I try to configure it, I see the following error:

$ tools/configure.sh -E qemu-intel64:nsh
[misc]
make[1]: Leaving directory '/home/jon/nuttx-test/nuttx'
Kconfig:2026: can't open file "/Kconfig"
make: *** [tools/Unix.mk:607: olddefconfig] Error 1
ERROR: failed to refresh

patacongo, 2022-04-02:
Two tips:

1. Do `sh -x tools/configure.sh -E qemu-intel64:nsh' to see what is going on.
2. Look at line 2026 of Kconfig:

2026 source "$APPSDIR/Kconfig"

CONCLUSION:  $APPSDIR is not defined.

Also in Kconfig:

   8 config APPSDIR
   9         string
  10         option env="APPSDIR"

patacongo, 2022-04-02:
The rest of the story is in tools/Unix.mk:

$(Q) APPSDIR=${CONFIG_APPS_DIR} EXTERNALDIR=$(EXTERNALDIR) kconfig-mconf Kconfig

3. Look at your .config file.  What did you select for CONFIG_APPS_DIR.  configure.sh will look for the apps/ directory and try to locate it for you, but possibly it did not find it.

dosadi, 2022-04-02:
Sorry, my bad - I had the apps directory in the wrong spot.  Moving it fixed the problem....

closed 2022-04-02 by dosadi