apache/nuttx #5973: CONFIG_NET_TCP_WRITE_BUFFERS=y lead to deadlock

pkarashchenko, 2022-04-03:
Recently I've been testing FTPD operation on custom SAME70 based board with FileZilla FTP client. I had `CONFIG_NET_TCP_WRITE_BUFFERS=y`. I have a folder with 2 files that I try to upload from laptop to device. Upload of a folder fails. FileZilla starts 2 FTP sessions in parallel one for each file upload so 2 instances of `ftpd_worker` are started. The situation happens in 80% of my tests. The FTP client send `"STOR"` command for both files and by adding extra logs I see that both threads reach `ftpd.c:1908` code `ret = ftpd_response(session->cmd.sd, session->txtimeout, g_respfmt1, 150, ' ', "Opening data connection");` at the same time so `send` is called concurrently. The logs before `ftpd_response` call are printed, but there are no logs after `ftpd_response` call. NSH "ps" shows that both threads are "waiting for semaphore", but most probably loop somewhere inside the `psock_tcp_send` trying to allocate `iob`. Setting `CONFIG_NET_TCP_WRITE_BUFFERS=n` solves the issue and FTP server operates as expected.

fjpanag, 2022-11-26:
I am hitting the exact same issue with a custom server, on high load.

TCP deadlocks during send when it tries to allocate an iob.

anchao, 2023-01-30:
Does this issue still exist on the latest nuttx? recently I fixed a similar issue, I don’t know if there is any improvement for ftpd

https://github.com/apache/nuttx/pull/8130

casaroli, 2026-11-08:
Possibly fixed by #8130.

closed 2026-11-08 by casaroli