apache/nuttx #6037: thread and task behavior should be consistent?

anjiahao1, 2022-04-11:
```c
#include <pthread.h>
#include <stdio.h>
#include <errno.h>

static void *a_thread_func()
{
	printf("123456\n");
	pthread_exit(0);
	return NULL;
}

int main(void)
{
	pthread_t new_th;
	pthread_attr_t new_attr;
	int ret_val;

	/* Initialize attribute */
	if (pthread_attr_init(&new_attr) != 0) {
		printf("Cannot initialize attribute object\n");
		return 1;
	}

	/* Set the attribute object to PTHREAD_CREATE_DETACHED. */
	if (pthread_attr_setdetachstate(&new_attr, PTHREAD_CREATE_DETACHED) !=
	    0) {
		printf("Error in pthread_attr_setdetachstate()\n");
		return 1;
	}

	/* Create a new thread passing it the new attribute object */
	if (pthread_create(&new_th, &new_attr, a_thread_func, NULL) != 0) {
		printf("Error creating thread\n");
		return 1;
	}

	/* If pthread_join() or pthread_detach fail, that means that the
	 * test fails as well. */
	ret_val = pthread_join(new_th, NULL);

	if (ret_val != EINVAL) {
		printf("Test FAILED\n");
		return 1;
	}
  	printf("123\n");

	if (ret_val != EINVAL) {
		printf("Test FAILED\n");
		return 1;
	}
	printf("Test PASSED\n");
	pthread_exit(0);
}
```
it is a testcase,i try it use
`gcc hello_main.c -o test -lpthread` and then `./test` output is
```
123
Test PASSED
123456
```
Then i use it in nuttx `hello` , output is
```
123
Test PASSED
```
`123456` is miss,i debug `ptread_exit`code,`pthread_completejoin` filed,then call `exit` kill all threads.

anjiahao1, 2022-04-11:
> `gcc hello_main.c -o test -lpthread` -- is this tested on RT-Linux? What are the priorities of `main` vs `pthread`?

i use ubuntu 20.04

pkarashchenko, 2022-04-11:
And in NuttX case both `main` and `pthread` have the same priorities? From the first look it look like `a_thread_func` didn't have a chance to be scheduled.

anjiahao1, 2022-04-11:
> And in NuttX case both `main` and `pthread` have the same priorities? From the first look it look like `a_thread_func` didn't have a chance to be scheduled.

This case cannot be used directly on nuttx, it will cause crash, this should be the problem of nuttx, I will patch the last time later

anjiahao1, 2022-04-11:
> > And in NuttX case both `main` and `pthread` have the same priorities? From the first look it look like `a_thread_func` didn't have a chance to be scheduled.
> 
> This case cannot be used directly on nuttx, it will cause crash, this should be the problem of nuttx, I will patch the last time later

[pr6038](https://github.com/apache/incubator-nuttx/pull/6038) will fix it

patacongo, 2022-04-11:
This might be correct behavior for NuttX.  In NuttX, the scheduler and priority are determined by the parent task.  So the pthread inherits SCHED_FIFO (probably) and same priority as the parent task.  Since the parent never gave the child an opportunity to run, the output from the pthread never occurred.

Linux works differently.  The scheduler will default to SCHED_OTHER and priority 5 (I think).  That will be inherited by the pthread (unless PTHREAD_EXPLICIT is used).  SCHED_OTHER is a fair scheduler, SCHED_FIFO is not and from my understanding of the FIFO scheduler, the behavior you show is correct.

What you are suggesting is that pthread_create() should yield.  I don't think would be correct.

In order to do a correct comparison with Linux, you would need to set SCHED_FIFO and an appropriate priority.  I would be interested in the result.  I don't know what would happen!!!

xiaoxiang781216, 2022-06-10:
Close, already fixed by #6038

closed 2022-06-10 by xiaoxiang781216