apache/nuttx #6112: nx_mkfifo buffer size

cburlacu, 2022-04-20:
It seems that the fifo buffer is loosing 1 byte:
```
Pipe has 127 free bytes
Successfuly written 32 bytes, free 95
Successfuly written 32 bytes, free 63
Successfuly written 32 bytes, free 31
??Written 31 bytes / 0
```
source code:

```
  char buff[32];
  int fd, count;
  nx_mkfifo(pipe_name, 0644, 128);
  fd = open(pipe_name, O_RDWR | O_NONBLOCK);
  ret = ioctl(fd, FIONSPACE, (unsigned long)&count);
  printf("Pipe has %d free bytes\n", count);

  while ((ret = write(fd, buff, sizeof(buff))) == sizeof(buff))
  {
    ioctl(fd, FIONSPACE, (unsigned long)&count);
    printf("Successfuly written %d bytes, free %d\n", ret, count);
  }
  printf("??Written %d bytes / %d\n", ret, errno);
```

xiaoxiang781216, 2022-04-21:
This is because the last space is used by implementation to indicate the fifo is full.

xiaoxiang781216, 2022-04-21:
@cburlacu could you try #6113 ?

cburlacu, 2022-04-21:
@xiaoxiang781216 Yes, it works:
```
Pipe has 128 free bytes
Successfuly written 32 bytes, free 96
Successfuly written 32 bytes, free 64
Successfuly written 32 bytes, free 32
Successfuly written 32 bytes, free 0
??Written -1 bytes / 11
```
Thanks!

closed 2022-04-21 by pkarashchenko