apache/nuttx #6219: Makefile does not detect new toolchain

kassane, 2022-05-08:
Ref: #5732 

When trying last to commit the proposal above there was an issue during the make build running the toolchain that its initialization was not listed in the makefile log. I expected at least some compilation error but it doesn't even return. Except when replacing the `CC` and `CXX` with `ZCC` and `ZCXX` there are some compilation errors on `Config.mk` file.

Would any further changes be required to add a new toolchain or did I forget some reference point?

**output:**

```bash
# [...]
CC:  signal/sig_ppoll.c
make[3]: Entrando no diretório '/home/kassane/Code/nuttxspace/apps/examples/hello_zig'

CC:  driver/fs_closeblockdriver.c
CC:  signal/sig_fillset.c
CC:  signal/sig_hold.c
CC:  signal/sig_pselect.c
CC:  driver/fs_blockpartition.c
CC:  signal/sig_ignore.c
CC:  task/task_create.c
CC:  signal/sig_ismember.c
CC:  driver/fs_findmtddriver.c
CC:  signal/sig_pause.c
CC:  task/task_init.c
CC:  driver/fs_blockproxy.c
CC:  signal/sig_psignal.c
CC:  task/task_setup.c
CC:  dirent/fs_closedir.c
CC:  signal/sig_raise.c
AR (add): libapps.a           
# Missing ZIG: examples/hello_zig_main.zig
make[3]: Saindo do diretório '/home/kassane/Code/nuttxspace/apps/examples/hello_zig'
# [...]
```

xiaoxiang781216, 2022-05-12:
Does the new toolchain on your search path?

kassane, 2022-05-12:
Right, the example of this is zcc and zcxx worked even though they generated a flags error. Only zig build did not return any information in the output.
I will review the `application.mk` file again.

The strange thing is that when did an external test manually everything works.
But only note in question is the condition to always link the object generated by zig to libc.

**What do you mean?**

zig code doesn't need libc (only ffi) and the other toolchains can't find to call it.

kassane, 2022-05-25:
Wow! Thx @lupyuen 

Ref.: https://github.com/lupyuen/zig-bl602-nuttx

closed 2023-01-11 by kassane