apache/nuttx #6507: raspiberrypi-pico reporting Mass Storage error after nuttx.uf2 was flashed

acassis, 2022-06-23:
I deviced to test raspberrypi-pico with the displaypack:

$ ./tools/configure.sh raspberrypi-pico:displaypack
$ make -j

Then after copying the nuttx.uf2 to it I got this report:

```
[31134.669325] usb 3-3: new full-speed USB device number 33 using xhci_hcd
[31134.819081] usb 3-3: New USB device found, idVendor=2e8a, idProduct=0003, bcdDevice= 1.00
[31134.819092] usb 3-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[31134.819096] usb 3-3: Product: RP2 Boot
[31134.819098] usb 3-3: Manufacturer: Raspberry Pi
[31134.819100] usb 3-3: SerialNumber: E0C912DXXXXX
[31134.823039] usb-storage 3-3:1.0: USB Mass Storage device detected
[31134.823436] scsi host0: usb-storage 3-3:1.0
[31135.842518] scsi 0:0:0:0: Direct-Access     RPI      RP2              1    PQ: 0 ANSI: 2
[31135.842958] sd 0:0:0:0: Attached scsi generic sg0 type 0
[31135.844030] sd 0:0:0:0: [sda] 262144 512-byte logical blocks: (134 MB/128 MiB)
[31135.844737] sd 0:0:0:0: [sda] Write Protect is off
[31135.844742] sd 0:0:0:0: [sda] Mode Sense: 03 00 00 00
[31135.845666] sd 0:0:0:0: [sda] No Caching mode page found
[31135.845689] sd 0:0:0:0: [sda] Assuming drive cache: write through
[31135.885629]  sda: sda1
[31135.903523] sd 0:0:0:0: [sda] Attached SCSI removable disk
[31136.451898] usb 3-3: USB disconnect, device number 33
[31136.452270] sd 0:0:0:0: [sda] tag#0 FAILED Result: hostbyte=DID_ERROR driverbyte=DRIVER_OK cmd_age=0s
[31136.452280] sd 0:0:0:0: [sda] tag#0 CDB: Read(10) 28 00 00 00 03 08 00 00 f0 00
[31136.452283] blk_update_request: I/O error, dev sda, sector 776 op 0x0:(READ) flags 0x84700 phys_seg 30 prio class 0
[31136.452321] blk_update_request: I/O error, dev sda, sector 1016 op 0x0:(READ) flags 0x80700 phys_seg 1 prio class 0
[31136.452449] blk_update_request: I/O error, dev sda, sector 776 op 0x0:(READ) flags 0x0 phys_seg 1 prio class 0
[31136.452464] Buffer I/O error on dev sda, logical block 97, async page read
```

acassis, 2022-06-23:
Seams like NuttX is freezing during the start-up. I connected a USB/Serial on GPIO0/GPIO1 and only got:
```
ACE
```

I am using this config now:
```
$ ./tools/configure.sh raspberrypi-pico:waveshare-lcd-1.14
```

yunkya2, 2022-06-26:
Hi, @acassis 
The mass storage error seems not related to the issue.
I found that the change 9140693 is the root cause. By reverting this change, I confirmed raspberrypi-pico:displaypack works.
I have not investigated what this change does and what is wrong yet.

yunkya2, 2022-06-26:
I found the problem and fixed in #6526.
Now I confirmed `raspberrypi-pico:displaypack` works fine.
I think waveshare-lcd-1.14 also works because it has same LCD controller.

closed 2022-06-26 by pkarashchenko

acassis, 2022-06-26:
Thank you very much @yunkya2