apache/nuttx #6531: IMXRT USB speed may need to be forced to FULL when CONFIG_USBDEV_DUALSPEED is not configured

edsut, 2022-06-27:
I am working with a Teensy4.1 and had a consistent hardfault whenever I connected QGC to Mavlink over USB.  I eventually realized that I did not have CONFIG_USBDEV_DUALSPEED enabled, but line 1975 of imxrt_usbdev.c (https://github.com/apache/incubator-nuttx/blob/master/arch/arm/src/imxrt/imxrt_usbdev.c#L1975) was being executed at startup.  The iMXRT1062 (Teensy4.1 SOC) does support HIGH speed, so I enabled CONFIG_USBDEV_DUALSPEED and the crash disappeared.
There should be some coordination between CONFIG_USBDEV_DUALSPEED and the link's speed at startup.
I modified imxrt_wakeup to force FULL speed if CONFIG_USBDEV_DUALSPEED is not set, and this allowed my code to avoid the hardfault regardless of the setting of CONFIG_USBDEV_DUALSPEED...
```
static int imxrt_wakeup(struct usbdev_s *dev)
{
  irqstate_t flags;

  usbtrace(TRACE_DEVWAKEUP, 0);

  flags = enter_critical_section();
#ifndef CONFIG_USBDEV_DUALSPEED
  imxrt_setbits(USBDEV_PRTSC1_PTC_FS | USBDEV_PRTSC1_FPR, IMXRT_USBDEV_PORTSC1);
#else
  imxrt_setbits(USBDEV_PRTSC1_FPR, IMXRT_USBDEV_PORTSC1);
#endif
  leave_critical_section(flags);
  return OK;
}

Any thoughts on this?
```

edsut, 2022-06-28:
LOL!  I wish I did have a friend at NXP!  I was just posting an update as you added your comment...
After a few days of testing, my crash is not resolved.  Prior to enabling CONFIG_USBDEV_DUALSPEED, I was not able to do any sensor calibration through QGC (because of the crash); then after enabling it I was able to calibrate (accelerometer)  so I was off on a tangent for a few days.  Now, I am seeing that it just crashes at a different point (still related to USB/QGC somehow).  
I also added a line of code to imxrt_usbreset() that will keep line 1975 from being called when CONFIG_USBDEV_DUALSPEED is not enabled, but at this point I don't know if that has any value (to include a possible negative effect).  Bottom line... I'm back where I started, still trying to understand this crash.

acassis, 2022-06-30:
@edsut did you try to ask help from PX4 community as well since it is also related to QGC? I think there are more people using it and they will help to validate if your modification. After that I think it could submit a PR here with the right solution. I have iMXRT board but I never used QGC before.

davids5, 2022-06-30:
I am ooo til 7-5. I will look at it ASAP when I return

edsut, 2022-07-01:
@acassis  I did post the question there prior to this... https://discuss.px4.io/t/mavlink-hardfault-on-usb-connection/27856
No response.
My gut says its not QGC specific (my hunch is still that it is deep in the USB driver).
Prior to enabling CONFIG_USBDEV_DUALSPEED, I was able to cause the crash with a snippet of code that just wrote a large chunk of data to USB (see https://discuss.px4.io/t/possible-bug-in-imxrt-usb-interface/28005).  Now (with CONFIG_USBDEV_DUALSPEED enabled, that reproduction snippet doesn't crash, so at the moment my only way to reproduce it is with certain interactions with QGC.
One new bit of information...
I wrote a Linux program that monitors the connection between QGC (on laptop) and my Teensy4.1.  With that inserted between the two devices the above crash does NOT occur!!  Clearly with that inserted, buffering is different from end-to-end, but data (messages sent between the two) should be identical.

edsut, 2022-07-05:
The details of this issue are scattered (my fault) across a few forum posts and github issues so I'll try to recap here...
Note in all cases, "crash" refers to a hardfault-CPU-crash, not a physical crash of a drone...

I first noticed a consistent crash while doing calibration tests with QGC and PX4 (running Mavlink over USB).  Unfortunately the address of the fault was inconsistent, so it was not an easy event to trace back to.  Eventually I found I was able to reproduce the crash by simply writing a large block of data to the USB interface running just NuttX (i.e.  this is not a PX4/Mavlink problem).  After more testing/analysis I discovered what I thought might be the problem (refer to my initial post in this issue), so I enabled CONFIG_USBDEV_DUALSPEED, and after that I was not able to cause the crash by writing a large block of data to USB... While I think that is a valid change, after further investigation, it appears that the bug is still lurking.   I know this because now I can consistently cause a crash at a different point in my QGC<->PX4 interaction.  Apparently enabling CONFIG_USBDEV_DUALSPEED just caused the bug to show itself in a different way.

All that being said, the easiest way to reproduce this problem is to leave CONFIG_USBDEV_DUALSPEED disabled and write a large (1000 bytes) block of data to USB.  I've reproduced the problem with NuttX-10.2.0 and NuttX-10.3.0 (master).  While I can reproduce the crash with the standard **teensy-4.x:nsh-4.1** configuration, that doesn't show the dump of the hardfault.  I modified the configuration to run the console on a UART so that the USB driver was not involved with the console.  In this case the dump from the hardfault-crash is shown on the UART-console (but doesn't show a useful backtrace).  
All it takes is this code (I just created an example app to invoke this)...

```
int fd = open("/dev/ttyACM0", O_RDWR | O_NOCTTY));
write(fd,(uint8_t *)0x20200000,1000);
close(fd);
```
Note that this code ONLY causes the crash when CONFIG_USBDEV_DUALSPEED is not enabled.  When I enable CONFIG_USBDEV_DUALSPEED, the reproduction scenario gets a bit more complicated; however I believe they are one-and-the same bug.  If anyone has a clue (I have yet to dig deep into the guts of iMXRT1062 USB/Teensy4.1 driver; but I suspect that is where the problem lies)

edsut, 2022-07-05:
More information...
Thanks to suggestion by @acassis, I used the usbserial application to cause the crash.  
Its tough to work with unless a new defconfig is used to allow a UART to run as the console.
Note also that when I enabled CONFIG_USBDEV_DUALSPEED, then usbserial did not crash.
@davids5 any thoughts on this?

edsut, 2022-07-06:
More information:
I now have DCACHE turned off and CONFIG_USBDEV_DUALSPEED enabled.  Crash still occurs...
Running usbserial app, consistently takes a hardfault, and I believe the value of privep->head at line [908 of imxrt_usbdev.c](https://github.com/apache/incubator-nuttx/blob/master/arch/arm/src/imxrt/imxrt_usbdev.c#L908) has a corrupt pointer at the time of the crash.  I will be OOO from now till Monday.  Will continue analysis then.

edsut, 2022-07-13:
@davids5  I've got to be getting close here...
I now see that just prior to the hardfault, the value of [bytesleft in imxrt_usbdev.c](https://github.com/apache/incubator-nuttx/blob/master/arch/arm/src/imxrt/imxrt_usbdev.c#L953) is negative.  I'll start digging into that, but if you have any thoughts,  I'd sure appreciate it...

edsut, 2022-07-19:
I've pulled in NuttX10.3.0's imxrt_usbdev.c code, and so far (two days of testing) it has not hard-faulted.  I -thought- I had done this a few weeks back, but apparently I didn't so I'll close this issue.

closed 2022-07-19 by edsut