apache/nuttx #6569: Why RISC-V up_testset use lr/sc and not amoswap?

pkarashchenko, 2022-07-04:
The RISC-V supports "Atomic Memory Operations" instruction. Seems like up_testset can be implemented with the less number of instruction using `amoswap.w.aq`/`amoswap.w.rl` than current implementation that use `lr`/`sc` instructions.

xiaoxiang781216, 2022-07-04:
The implementation is same as you want:
https://github.com/apache/incubator-nuttx/blob/master/arch/risc-v/src/common/riscv_testset.S#L34-L40

pkarashchenko, 2022-07-04:
@xiaoxiang781216 the issue is that currently it is implemented as you pointed, but it use `lr`/`sc`, but I posted an issue that we can rework it to use `amoswap`.
Current code:
```
  .globl up_testset
  .type  up_testset, %function

up_testset:

  li       a1, SP_LOCKED

  /* Test if the spinlock is locked or not */

retry:
  LR_INST  a2, (a0)       /* Test if spinlock is locked or not */
  beq      a2, a1, locked /* Already locked? Go to locked: */

  /* Not locked ... attempt to lock it */

  SC_INST  a2, a1, (a0)   /* Attempt to set the locked state (a1) to (a0) */
  bnez     a2, retry      /* a2 will not be zero, if sc.b failed, try again */

  /* Lock acquired -- return SP_UNLOCKED */

  fence                   /* Required before accessing protected resource */
  li        a0, SP_UNLOCKED
  jr        ra

  /* Lock not acquired -- return SP_LOCKED */

locked:
  li        a0, SP_LOCKED
  jr        ra
  .size	up_testset, . - up_testset
  .end
```
But can be something like
```
  .globl up_testset
  .type  up_testset, %function

up_testset:

  li       a1, SP_LOCKED
  amoswap.w a2, a1, (a0) /* Attempt to acquire spinlock atomically */
  beq      a2, a1, locked /* Already locked? Go to locked: */

  /* Lock acquired -- return SP_UNLOCKED */

  fence                   /* Required before accessing protected resource */
  li        a0, SP_UNLOCKED
  jr        ra

  /* Lock not acquired -- return SP_LOCKED */

locked:
  li        a0, SP_LOCKED
  jr        ra
  .size	up_testset, . - up_testset
  .end
```

xiaoxiang781216, 2022-07-04:
Sorry, I misunderstand your meaning.

pkarashchenko, 2022-07-04:
Maybe my suggestion has some drawbacks compared to current implementation so I'm posting this as an issue so people some discussion may happen

masayuki2009, 2022-07-05:
@pkarashchenko 
I remember I did the initial implementation for K210 SMP.
Actually, I referred to the armv7-a implementation that's why lr/sc are used for it.
However, I think `amoswap` instruction should work for up_testset.

PR #12232 (2024-04-25): arch/riscv: replace atomic operations to AMO
## Summary

arch/riscv: replace atomic operations to AMO

RISC-V provided fetch-and-op style atomic primitives as they scale to highly parallel systems better than LR/SC or CAS. A simple microarchitecture can implement AMOs using the LR/SC primitives, provided the implementation can guarantee the AMO eventually completes. More complex implementations might also implement AMOs at memory controllers, and can optimize away fetching the original value when the destination is x0.

https://github.com/riscv/riscv-isa-manual/blob/main/src/a-st-ext.adoc

##  [... cut]
closed 2024-04-25 by masayuki2009