apache/nuttx #6624: PANIC on TIM7 IRQ in timer example on armv6-m without FRAME_POINTERS

ALTracer, 2022-07-15:
## Summary
I get consistent crashes while testing apps/examples/timer with disabled CONFIG_FRAME_POINTERS.
With them enabled, it works, but .text consumes +4.5KiB flash.

## Impact
armv6-m thumb MCUs using basic timer peripherals.

## Testing
stm32f072b-disco with custom config, Milandr mdr32f1qi with custom config.

## Details

After porting basic timer functionality from stm32f0l0g0 onto mdr32, I tried to launch some userspace demos. It worked from time to time. It crashed when I tried to reduce consumed flash space. I even enabled tmrinfo debug prints, but they didn't help much in case of crashes -- though they showed successfull signal passing when everything worked.

I think I found a single Kconfig option that breaks it: CONFIG_FRAME_POINTERS=n. It changes CFLAGS from `-fno-omit-frame-pointers -fno-optimize-sibling-calls` to `-fomit-frame-pointers` and saves space. I couldn't get anything else to break at runtime.

Stackdumps/backtraces, defconfigs and workdir patches will follow. I used release/10.3 branch.

ALTracer, 2022-07-15:
stm32f072-disco:btimer defconfig diff with stm32f072-disco:nsh defconfig:
I enabled POSIX_TIMERS, PTHREAD, BUILTIN, EXAMPLES_TIMER, STM32F0L0G0_TIM7, TIMER, TIMER_ARCH.

Board timer initialization:
```diff
diff --git a/boards/arm/stm32f0l0g0/stm32f072-discovery/src/stm32_bringup.c b/boards/arm/stm32f0l0g0/stm32f072-discovery/src/stm32_bringup.c
index f52f9d545..fb9981ba4 100644
--- a/boards/arm/stm32f0l0g0/stm32f072-discovery/src/stm32_bringup.c
+++ b/boards/arm/stm32f0l0g0/stm32f072-discovery/src/stm32_bringup.c
@@ -30,6 +30,11 @@
 
 #include <syslog.h>
 
+#ifdef CONFIG_TIMER
+#include <nuttx/irq.h>
+#include "stm32_tim.h"
+#endif
+
 #include "stm32f072-discovery.h"
 
 #ifdef CONFIG_EXAMPLES_MODBUS
@@ -80,6 +85,17 @@ int stm32_bringup(void)
   }
 #endif
 
+#if defined(CONFIG_TIMER)
+  /* Initialize the timer, at this moment it's only Basic timers 6,7 */
+
+  #if defined(CONFIG_STM32F0L0G0_TIM6)
+    ret = stm32_timer_initialize("/dev/timer6", 6);
+  #endif
+  #if defined(CONFIG_STM32F0L0G0_TIM7)
+    ret = stm32_timer_initialize("/dev/timer7", 7);
+  #endif
+#endif
+
   UNUSED(ret);
   return OK;
 }
```
Hardware header aborting the build:
```diff
diff --git a/arch/arm/src/stm32f0l0g0/hardware/stm32_tim.h b/arch/arm/src/stm32f0l0g0/hardware/stm32_tim.h                                                                                                                           [170/593]
index ceb057d57..8d69c3024 100644                                   
--- a/arch/arm/src/stm32f0l0g0/hardware/stm32_tim.h                 
+++ b/arch/arm/src/stm32f0l0g0/hardware/stm32_tim.h                 
@@ -26,7 +26,7 @@   
  ****************************************************************************/
                    
 #if defined(CONFIG_STM32F0L0G0_STM32F0) || defined(CONFIG_STM32F0L0G0_STM32L0)
-#error "Timer definitions should be validated for target before use"
+#warning "Timer definitions should be validated for target before use"
 #endif                
                                 
 /* TODO Missing TIM2 definitions available on STM32G0x1 */
@@ -1101,7 +1101,9 @@            
 #define BTIM_CR1_URS              (1 << 2)  /* Bit 2: Update Request Source */
 #define BTIM_CR1_OPM              (1 << 3)  /* Bit 3: One Pulse Mode */
 #define BTIM_CR1_ARPE             (1 << 7)  /* Bit 7: Auto-Reload Preload enable */
+#if defined(CONFIG_STM32F0L0G0_STM32G0)
 #define BTIM_CR1_UIFREMAP         (1 << 11) /* Bit 11: UIF status bit remapping */
+#endif                                                                 
                                                                        
 /* Control register 2 */                                               
                                       
@@ -1126,6 +1128,8 @@                               
                                                    
 /* Counter register */                 
                                                                    
+#if defined(CONFIG_STM32F0L0G0_STM32G0)            
 #define BTIM_CNT_UIFCPY           (1 << 31)  /* Bit 31: UIF copy */
+#endif                                                      
                                                                                                                                            
 #endif /* __ARCH_ARM_SRC_STM32F0L0G0_HARDWARE_STM32_TIM_H */
```
Basic timers work fine, and I don't need the rest yet.

I also have backported DEBUG_OPT_UNUSED_SECTIONS and ARCH_HAVE_BACKTRACE for arm_backtrace_thumb.
There are some autoleds- and pinmapping-related changes in my board/stm32f072-disco directory, I made them because it's not stm32f051-disco board and I wanted to test Modbus slave on a second UART. That works fine.

ALTracer, 2022-07-15:
stm32f072-disco crash log, stackdump and backtrace:
```
ABCDF

NuttShell (NSH) NuttX-10.3.0
nsh> uname -a
NuttX 10.3.0 098866c45-dirty Jul 15 2022 15:38:41 arm stm32f072-discovery
nsh> free
                   total       used       free    largest  nused  nfree
        Umem:      12688       6920       5768       5496     45      2
nsh> ps
  PID GROUP PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK   STACK   USED  FILLED    CPU COMMAND
    0     0   0 FIFO     Kthread N-- Ready              00000000 000760 000384  50.5%   98.9% Idle Task
    2     1 100 RR       pthread --- Waiting  Semaphore 00000000 001024 000608  59.3%    0.0% mbs_poll 0
    3     3 100 RR       Task    --- Running            00000000 001768 001328  75.1%    2.1% nsh_main
nsh> modbus -s
modbus_main: Protocol stack is running
nsh> modbus -d
nsh> modbus -s
modbus_main: Protocol stack is stopped
nsh> free ps
nsh: free: too many arguments
nsh> free
                   total       used       free    largest  nused  nfree
        Umem:      12688       5368       7320       4264     39      6
nsh> ps
  PID GROUP PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK   STACK   USED  FILLED    CPU COMMAND
    0     0   0 FIFO     Kthread N-- Ready              00000000 000760 000384  50.5%   98.5% Idle Task
    3     3 100 RR       Task    --- Running            00000000 001768 001328  75.1%    2.2% nsh_main
nsh> help
help usage:  help [-v] [<cmd>]

  ?        date     exec     help     kill     mount    sleep    usleep   
  cat      df       exit     hexdump  ls       printf   time     xd       
  cp       echo     free     irqinfo  mkdir    ps       uname    

Builtin Apps:
  timer   sh      hello   nsh     modbus  
nsh> ls -l /dev
/dev:
 crw-rw-rw-       0 console
 crw-rw-rw-       0 null
 crw-rw-rw-       0 timer6
 crw-rw-rw-       0 timer7
 crw-rw-rw-       0 ttyS0
 crw-rw-rw-       0 ttyS1
nsh> timer
Open /dev/timer0
ERROR: Failed to open /dev/timer0: 2
nsh> timer -d /dev/timer7
Open /dev/timer7
  flags: 00000000 timeout: 1365 timeleft: 65520 nsignals: 0
Set timer interval to 1000000
  flags: 00000000 timeout: 999420 timeleft: 65534 nsignals: 0
Attach timer handler
  flags: 00000002 timeout: 999420 timeleft: 65501 nsignal[14] arm_hardfault: 
Hard Fault:
[14] arm_hardfault:   PC: 0x4000140a INSN: 0000
[14] arm_hardfault:   IRQ: 3 regs: 0x20002d70
[14] arm_hardfault:   PRIMASK: 00000000 IPSR: 00000003
[14] arm_hardfault: PANIC!!! Hard fault
[14] up_assert: Assertion failed at file:armv6-m/arm_hardfault.c line: 124 task: timer
[14] backtrace:
[14] [14] [<0x800f452>] up_backtrace+0xa/0x12c
[14] [14] [<0x800f096>] sched_backtrace+0x20/0x28
[14] [14] [<0x800cd9c>] sched_dumpstack+0x10/0x50
[14] [14] [<0x8003c92>] up_assert+0x46/0x234
[14] [14] [<0x8002e94>] _assert+0x2/0xe
[14] [14] [<0x8000c88>] arm_hardfault+0x74/0xa4
[14] [14] [<0x8001162>] irq_dispatch+0x46/0x98
[... 61 lines cut ...]
[14] [14] [<0x800ec1e>] timer_main+0x146/0x220
[14] [14] [<0x8002fd8>] nxtask_startup+0x10/0x16
[14] [14] [<0x8001616>] nxtask_start+0x42/0x4c
[14] backtrace:
[14] [ 3] [<0x8004220>] arm_switchcontext+0xc/0x10
[14] [ 3] [<0x8003ee0>] up_block_task+0x60/0x70
[14] [ 3] [<0x8003edc>] up_block_task+0x5c/0x70
[14] [ 3] [<0x800bd12>] nxsem_wait+0x3e/0x5c
[14] [ 3] [<0x800bc6c>] nx_waitpid+0x74/0x80
[14] [ 3] [<0x800bc7a>] waitpid+0x2/0x1a
[14] [ 3] [<0x8007c98>] nsh_builtin+0x50/0xdc
[14] [ 3] [<0x8005edc>] nsh_parse_command+0x14c/0x394
[14] [ 3] [<0x8008d72>] file_write+0x1a/0x28
[14] [ 3] [<0x8008da8>] write+0x2/0x1a
[14] [ 3] [<0x8005a5e>] readline_putc+0x10/0x2a
[14] [ 3] [<0x8006148>] nsh_parse+0x24/0x78
[14] [ 3] [<0x80059b4>] nsh_session+0x84/0x108
[14] [ 3] [<0x80056ba>] nsh_consolemain+0x14/0x22
[14] [ 3] [<0x8004958>] nsh_main+0x28/0x44
[14] [ 3] [<0x8002fd8>] nxtask_startup+0x10/0x16
[14] backtrace:
[14] [ 3] [<0x8004220>] arm_switchcontext+0xc/0x10
```
Notice it tries to jump into peripheral region ($pc=0x4000140a), that's TIM7 0x40001400-0x400017ff. 
Same for TIM6 -- $pc=0x4000100a, peripheral region is 0x40001000-0x400013ff.

ALTracer, 2022-07-15:
Milandr mdr32f1qi stackdump/backtrace:
```
nsh> ps
  PID GROUP PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK   STACK   USED  FILLED COMMAND
    0     0   0 FIFO     Kthread N-- Ready              00000000 001000 000368  36.8%  Idle Task
    1     1 100 RR       Task    --- Running            00000000 002024 001344  66.4%  nsh_main
nsh> free
                   total       used       free    largest  nused  nfree
        Umem:      43936       7120      36816      32752     48      5
nsh> timer
Open /dev/timer1
  flags: 00000000 timeout: 0 timeleft: 0 nsignals: 0
Set timer interval to 100
  flags: 00000000 timeout: 100 timeleft: 100 nsignals: 0
Attach timer handler
  flags: 00000002 timeout: 100 timeleft: 100 nsignals: 0
Start the timer
  flags: 00000003 timeout: 100 timeleft: 93 nsignals: 0
  flags: 00000003 timeout: 100 timeleft: 85 nsignals: 0
  flags: 00000003 timeout: 100 timeleft: 77 nsignals: 0
  flags: 00000003 timeout: 100 timeleft: 69 nsignals: 0
  flags: 00000003 timeout: 100 timeleft: 61 nsignals: 0
  flags: 00000003 timeout: 100 timeleft: 53 nsignals: 0
  flags: 00000003 timeout: 100 timeleft: 45 nsignals: 0
  flags: 00000003 timeout: 100 timeleft: 38 nsignals: 0
  flags: 00000003 timeout: 100 timeleft: 30 nsignals: 0
  flags: 00000003 timeout: 100 timeleft: 22 nsignals: 0
  flags: 00000003 timeout: 100 timeleft: 14 nsignals: 0
  flags: 00000003 timeout: 100 timeleft: 6 nsignals: 0
arm_hardfault: 
Hard Fault:
arm_hardfault:   PC: 0 INSN: 12f4
arm_hardfault:   IRQ: 3 regs: 0x201038e8
arm_hardfault:   PRIMASK: 00000001 IPSR: 00000003
arm_hardfault: PANIC!!! Hard fault
up_assert: Assertion failed at file:armv6-m/arm_hardfault.c line: 124 task: timer
backtrace:
[ 6] [<0x12466>] up_backtrace+0x12c0000000a/0x11
[ 6] [<0x11c8a>] sched_backtrace+0x2a00000020/0x11
[ 6] [<0xdeb8>] sched_dumpstack+0x5000000010/0x10
[ 6] [<0x4a3c>] up_assert+0x21400000040/0x10
[ 6] [<0x349a>] _assert+0xc00000002/0x10
[ 6] [<0xc00>] arm_hardfault+0xa400000074/0xf
[ 6] [<0xf5c>] irq_dispatch+0x480000001c/0xf
[ 6] [<0xb74>] arm_doirq+0x340000001c/0xf
[ 6] [<0x7e8>] exception_common+0x6800000024/0xf
[ 6] [<0xfffffffe>] g_timerops+0xfffe50effffe50ee/0x14
[ 6] [<0x2>] Unknown+0x2/0xd
[ 6] [<0x4c6c>] up_block_task+0x700000005c/0x10
[ 6] [<0xd828>] nxsig_timedwait+0xfc000000a0/0x10
[ 6] [<0xd580>] nxsig_nanosleep+0x800000002c/0x10
[ 6] [<0xd628>] clock_nanosleep+0x5c00000054/0x10
[ 6] [<0xee5a>] usleep+0x380000002a/0x10
backtrace:
[ 6] [<0x10490>] timer_main+0x21c00000164/0x11
[ 6] [<0x3624>] nxtask_startup+0x1600000010/0x10
[ 6] [<0x13aa>] nxtask_start+0x4c00000042/0x10
arm_registerdump: R0: 00000002 R1: 20102a78 R2: 201003cc  R3: 20102a78
arm_registerdump: R4: 20100354 R5: 20102a78 R6: 20100444  FP: 00000006
arm_registerdump: R8: 00000000 SB: 00000000 SL: 00000000 R11: 00000000
arm_registerdump: IP: 00000000 SP: 20103930 LR: 00004c71  PC: 00000002
arm_registerdump: xPSR: 20000238 PRIMASK: 00000001 CONTROL: 00000000
arm_dump_stack: User Stack:
arm_dump_stack: sp:     20103878
arm_dump_stack:   base: 20103248
arm_dump_stack:   size: 000007f0
arm_stackdump: 20103860: 00000001 00000000 00004b45 00017f69 20103878 00004b55 00000001 00000000
arm_stackdump: 20103880: 00000002 000040fd 00000000 000007f0 201038a8 00000000 00017aa8 201038e8
arm_stackdump: 201038a0: 00000003 0000349f 00000000 00000c05 00000003 201038e8 00000000 20100468
arm_stackdump: 201038c0: 20100348 00000003 00000006 00000f5f 00000000 00000b79 00000000 00000000
[... 43 lines cut ...]
[ 6] [<0x4972>] arm_dump_backtrace+0xc00000006/0x10
[ 6] [<0xd0c8>] nxsched_foreach+0x3c0000002c/0x10
[ 6] [<0x4b6a>] up_assert+0x2140000016e/0x10
[ 6] [<0x349a>] _assert+0xc00000002/0x10
[ 6] [<0xc00>] arm_hardfault+0xa400000074/0xf
[ 6] [<0xf5c>] irq_dispatch+0x480000001c/0xf
[ 6] [<0xb74>] arm_doirq+0x340000001c/0xf
[ 6] [<0x7e8>] exception_common+0x6800000024/0xf
[ 6] [<0xfffffffe>] g_timerops+0xfffe50effffe50ee/0x14
[ 6] [<0x2>] Unknown+0x2/0xd
[ 6] [<0x4c6c>] up_block_task+0x700000005c/0x10
[ 6] [<0xd828>] nxsig_timedwait+0xfc000000a0/0x10
[ 6] [<0xd580>] nxsig_nanosleep+0x800000002c/0x10
backtrace:
[ 6] [<0xd628>] clock_nanosleep+0x5c00000054/0x10
[ 6] [<0xee5a>] usleep+0x380000002a/0x10
[ 6] [<0x10490>] timer_main+0x21c00000164/0x11
[ 6] [<0x3624>] nxtask_startup+0x1600000010/0x10
[ 6] [<0x13aa>] nxtask_start+0x4c00000042/0x10
```
This time, $pc is at 0x0, which is start of Flash. 
TIMER1 base address on this platform is at 0x40007000, but it doesn't show up.
AHB-Lite SRAM at 0x20100000 as main memory, DTCM SRAM at 0x20000000 as OS heap (noexec nodma).

OS crashes in processing TIMER1 interrupt while trying to do the timer_notify callback to send a 17 signal to timer_main POSIX signal handler and switch tasks in up_block_task, (but g_readytorun.head->xcp.regs is bad? -- investigate). I can catch it with a hardware breakpoint at *0x0 and *0x2.

acassis, 2022-07-29:
@ALTracer did this also happens on STM32F1/F4 or only STM32F0 ?

acassis, 2022-07-29:
BTW, never heard about Milandr MDR32 before and there is not much info about it on Internet. Are you planing to add support to it on mainline?

ALTracer, 2022-07-30:
@acassis I could verify timers on STM32F429I-disco board sometime next week, though it's armv7e-m. I don't have access to any STM32F1-based boards.

Milandr has a line of MCUs produced since ~2010, see the official manufacturer site https://ic.milandr.ru/products/mikrokontrollery_i_protsessory/ and unofficial wiki http://startmilandr.codepx.ru/doku.php/prog:start (both sites in Russian).
Some of them are ARM Cortex-M1 or M3, some RISC-V RV32IMA.

The LDM-Systems company sells evaluation boards https://ldm-systems.ru/catalog/helper?cat=milandr&scat=op

I am not affiliated with them, we're only evaluating their chips for our next network-connected device with all silicon components originating from Russia. I thought about bringing mainline support, but I've been porting NuttX/10.3 for a month now, the only board is only 50% feature-complete, and I still rely on vendor's open-source (?) HAL for a couple of drivers. I will consult with my employer by the end of August 2022 if anything useful comes out of this project.