apache/nuttx #6641: CONFIG_DEBUG_MM=y breaks procfs output

pkarashchenko, 2022-07-20:
Having a config with `CONFIG_DEBUG_FEATURES=y` and `CONFIG_DEBUG_MM=y` enabled. When running `ps` getting the next output:
```
nsh> ps
  PID GROUP PRI POLICY   TYPE    NPX STATE    EVENT     SIGMASK      HEAP  STACK   USED  FILLED COMMAND
    0     0   0 FIFO     Kthread N-- Ready              00000000nsh: ps: read failed: 22
 00000000 001000 000408  40.8%  Idle Task
    1     1 224 RR       Kthread --- Waiting  Semaphore 00000000nsh: ps: read failed: 22
 00000000 001992 000276  13.8%  hpwork 0x204014f8
    2     2 100 RR       Kthread --- Waiting  Semaphore 00000000nsh: ps: read failed: 22
 00000000 001992 000548  27.5%  lpwork 0x20401508
    3     3 100 RR       Task    --- Running            00000000nsh: ps: read failed: 22
 00000000 002000 001436  71.8%  nsh_main
    4     4 224 RR       Kthread --- Waiting  Semaphore 00020000nsh: ps: read failed: 22
```

ALTracer, 2022-07-21:
This must be a recent change. I cannot confirm breakage, but release/10.3 behaves well for me, showing all the tasty runtime statistics including heap per-task.

I see your column HEAP shows zero sizes. Maybe something related to mallinfo?
Can you also check /proc/meminfo (`free`) and /proc/memdump? 
`nsh> echo used > /proc/memdump`

fjpanag, 2022-07-21:
I too observe this issue, and I just spent some time debugging it.

If you enable `CONFIG_MM_BACKTRACE`, then `ps` will work correctly.

The root cause is that heap information printing in `ps` is controlled by `CONFIG_DEBUG_MM`.  
This can be seen [here](https://github.com/apache/incubator-nuttx/blob/master/fs/procfs/fs_procfsproc.c#L93).

However, the actual implementation of this depends on `CONFIG_MM_BACKTRACE`.
You can see this [here](https://github.com/apache/incubator-nuttx/blob/master/fs/procfs/fs_procfsproc.c#L1578).

fjpanag, 2022-07-21:
Hmm, I think that the confusion between `CONFIG_DEBUG_MM` and `CONFIG_MM_BACKTRACE` is more widespread. Have a look [here](https://github.com/apache/incubator-nuttx/blob/master/include/nuttx/fs/procfs.h#L136) for example.

I think that all uses of `CONFIG_DEBUG_MM` need to be checked.

pkarashchenko, 2022-07-21:
Yes, `CONFIG_DEBUG_MM` and `CONFIG_MM_BACKTRACE` are messed up a bit.

fjpanag, 2022-07-21:
I am working on this, I will provide a fix soon.

closed 2022-07-21 by xiaoxiang781216