apache/nuttx #6642: Duplicate SNIOC_SET_INTERVAL in sensors/ioctl.h

lupyuen, 2022-07-20:
`SNIOC_SET_INTERVAL` is defined twice in [sensors/ioctl.h](https://github.com/apache/incubator-nuttx/blob/master/include/nuttx/sensors/ioctl.h#L174-L175) with different values.

`SNIOC_SET_INTERVAL` is first defined in [ioctl.h line 175](https://github.com/apache/incubator-nuttx/blob/master/include/nuttx/sensors/ioctl.h#L174-L175):

```c
/* SNIOC_READ_CONVERT_DATA */                     /* Arg: struct scd30_conv_data_s* */
#define SNIOC_SET_INTERVAL         _SNIOC(0x0054) /* Arg: uint16_t value (seconds) */
```

Then again in [ioctl.h line 251](https://github.com/apache/incubator-nuttx/blob/master/include/nuttx/sensors/ioctl.h#L246-L251) with a different value:

```c
/* Command:      SNIOC_SET_INTERVAL
 * Description:  Set interval between samples
 * Argument:     This is the interval pointer, in microseconds
 */
#define SNIOC_SET_INTERVAL         _SNIOC(0x0081)
```

Perhaps the first definition should be removed?

I discovered this when [converting NuttX Sensor Test to Zig](https://github.com/lupyuen/visual-zig-nuttx), and Zig incorrectly used the first value (instead of the correct second value). Thanks!

xiaoxiang781216, 2022-07-20:
@Donny9 could you take a look?

Donny9, 2022-07-20:
@lupyuen yes, we should remove the first definition and keep the cmd definition of common driver for SNIOC_SET_INTERVAL. And there are some other definitions (SNIOC_START,SNIOC_STOP...)that also have a duplication, i will submit a modification to remove them.

lupyuen, 2022-07-20:
@Donny9 Thank so you much, really appreciate it :-)