apache/nuttx #6737: Sensor Interval SNIOC_SET_INTERVAL: Pass by Value or Pointer?

lupyuen, 2022-07-29:
When setting the Sensor Interval, the Sensor Test App calls `ioctl(SNIOC_SET_INTERVAL)` passing a Pointer to the Interval:

https://github.com/apache/incubator-nuttx-apps/blob/master/testing/sensortest/sensortest.c#L328

```c
  // Pass Interval as Pointer
  ret = ioctl(fd, SNIOC_SET_INTERVAL, &interval);
```

But `SNIOC_SET_INTERVAL` expects a Value, not a Pointer:

https://github.com/apache/incubator-nuttx/blob/master/drivers/sensors/sensor.c#L580-L608

```c
static int sensor_ioctl(FAR struct file *filep, int cmd, unsigned long arg) {
      ...
      case SNIOC_SET_INTERVAL: {
          // Pass the Interval as Value (instead of Pointer)
          ret = sensor_update_interval(upper, user, arg);
```

https://github.com/apache/incubator-nuttx/blob/master/drivers/sensors/sensor.c#L200-L241

```c
// sensor_update_interval receives the Interval as Value
static int sensor_update_interval(FAR struct sensor_upperhalf_s *upper,
                                  FAR struct sensor_user_s *user,
                                  unsigned long interval) {
      ...
      // And passes Interval by Pointer to the driver
      ret = lower->ops->set_interval(lower, &min_interval);
```

The passing of Interval by Value vs Pointer was changed after this recent commit: https://github.com/apache/incubator-nuttx/commit/24040250f5d6645b4b1c0165d8a52323c755c7f6

Previously the Interval was passed by Pointer to `SNIOC_SET_INTERVAL`:

https://github.com/apache/incubator-nuttx/blob/ba5320c36f8eb978d799f89822f7e9d5dfdb63e7/drivers/sensors/sensor.c#L388-L401

I modified the Sensor Test App to pass by Value (instead of Pointer):

```c
  // Now we pass Interval by Value
  ret = ioctl(fd, SNIOC_SET_INTERVAL, interval);
```

And the [BME280 Driver](https://github.com/lupyuen/bme280-nuttx) now receives the correct Interval.

Could we clarify whether `SNIOC_SET_INTERVAL` should pass by Value or Pointer? Thanks!

Donny9, 2022-07-29:
> And the [BME280 Driver](https://github.com/lupyuen/bme280-nuttx) now receives the correct Interval.
> 
> Could we clarify whether `SNIOC_SET_INTERVAL` should pass by Value or Pointer? Thanks!

@lupyuen Sorry, i forget updating this cmd usage after https://github.com/apache/incubator-nuttx/commit/24040250f5d6645b4b1c0165d8a52323c755c7f6. we need to pass vlaue when using SNIOC_SET_INTERVAL and SNIOC_BATCH. The fix PR : https://github.com/apache/incubator-nuttx-apps/pull/1240, Thank you!

lupyuen, 2022-07-29:
Great thanks! :-)

closed 2022-07-29 by lupyuen