apache/nuttx #6914: Memory corruption on esp32s3

no1wudi, 2022-08-24:
On esp32s3-devkitc:nsh, enable UBSan and ostest, we can get the error report:
```
user_main: FPU test
Starting task FPU#1
fpu_test: Started tubsan_prologue: ================================================================================
ubsan_prologue: UBSAN: array-index-out-of-bounds in fpu.c:164:21
__ubsan_handle_out_of_bounds: index 80 is out of range for type 'fpu_threaddata_s [2]'
ubsan_epilogue: ================================================================================
ask FPU#1 at PID=4
FPU#1: pass 1
Starting task FPU#2
fpu_test: Started task FPU#2 at PID=5
FPU#81: pass 1
FPU#1: pass 2
FPU#81: pass 2
FPU#1: pass 3
FPU#81: pass 3
FPU#1: pass 4
FPU#81: pass 4
FPU#1: pass 5
FPU#81: pass 5
FPU#1: pass 6
FPU#81: pass 6
FPU#1: pass 7
FPU#81: pass 7
FPU#1: pass 8
FPU#81: pass 8
```
This `fpu_threaddata_s` defined in line 110:
```
static uint8_t g_fpuno;
/* static */ struct fpu_threaddata_s g_fputhread[FPU_NTHREADS];
```
The `g_funcno` was modified unexpectly.

gustavonihei, 2022-08-24:
Hi @no1wudi.
I've enabled **MM_UBSAN**, but it didn't report this `array-index-out-of-bounds` issue. Instead the execution simply crashed.
Could you please share the compiler version and the `defconfig` used for reproducing this UBSan error?

I've submitted #6917 fox fixing the crash, but the FPU#**X** counter is still incorrect:
```
user_main: FPU test                                                                                                                                                                                                                                                                       
Starting task FPU#1                                                                                                                                                                                                                                                                       
fpu_test: Started task FPU#1 at PID=4                                                                                                                                                                                                                                                     
FPU#1: pass 1                                                                                                                                                                                                                                                                             
Starting task FPU#2                                                                                                                                                                                                                                                                       
fpu_test: Started task FPU#2 at PID=5                                                                                                                                                                                                                                                     
FPU#1: pass 1                                                                                                                                                                                                                                                                             
FPU#1: pass 2                                                                                                                                                                                                                                                                             
FPU#1: pass 2                                                                                                                                                                                                                                                                             
FPU#1: pass 3                                                         
FPU#1: pass 3                                    
FPU#1: pass 4                                                         
FPU#1: pass 4                                                         
FPU#1: pass 5                                           
FPU#1: pass 5                                                         
FPU#1: pass 6                                                         
FPU#1: pass 6                                                         
FPU#1: pass 7                                                         
FPU#1: pass 7                                    
FPU#1: pass 8                                            
FPU#1: pass 8                                 
FPU#1: pass 9                                                         
FPU#1: pass 9                                                         
FPU#1: pass 10                                                        
FPU#1: pass 10                                                                                                                               
FPU#1: pass 11                                                        
FPU#1: pass 11                              
FPU#1: pass 12                                                        
FPU#1: pass 12                                                        
FPU#1: pass 13                                     
FPU#1: pass 13                                  
FPU#1: pass 14                                                        
FPU#1: pass 14                                                        
FPU#1: pass 15                                                        
FPU#1: pass 15                                                        
FPU#1: pass 16                                         
FPU#1: pass 16                                                        
FPU#1: Succeeded                                                      
FPU#1: Succeeded                          
fpu_test: Returning                                                   
                                                                      
End of test memory usage:                    
VARIABLE  BEFORE   AFTER                                              
======== ======== ========                                            
arena       64350    64350                                            
ordblks         3        7                                            
mxordblk    5fca0    5e680                                            
uordblks     4620     57b0                       
fordblks    5fd30    5eba0
```

no1wudi, 2022-08-25:
@gustavonihei CONFIG_MM_UBSAN_TRAP_ON_ERROR is enabled by default to reduce footprint, you can disable it to get complete error report instead of crash immediately while error detected.

PR #6917 (2022-08-25): ESP32-S3: Fix alignment of FPU registers in exception context
## Summary
This PR intends to fix the alignment of FPU registers in exception context for the **ESP32-S3** chip.
**ESP32-S3** requires a minimum alignment of 16 bytes for the total save area.

## Impact
Fixes #6914.

## Testing
Successful execution of `ostest` application on `esp32s3-devkit:nsh` with `MM_UBSAN` enabled.
closed 2022-08-25 by xiaoxiang781216