apache/nuttx #6938: Nuttx not installing correctly on the Particle Photon.

greg5678, 2022-08-27:
Here are the commands I've used. 
```
$ mkdir nuttxspace
$ cd nuttxspace
$ git clone https://github.com/apache/incubator-nuttx.git nuttx
$ git clone https://github.com/apache/incubator-nuttx-apps apps
$ cd nuttx
$ ./tools/configure.sh -m photon:wlan
$ make menuconfig
```

Another post said it better than I ever could. 
"Application Configuration → Network Utilities → Network initialization → WAPI Configuration
set SSID and Passphrase corresponding to the Wi-Fi AP (Router)
save, exit"

```
$ make
$ sudo dfu-util -d 2b04:d006 -a 0 -s 0x08020000 -D nuttx.bin
```
The device just continues to blink yellow and if I unplug neither the status light nor the D7 LED comes on, but I can still put the device in DFU mode (which causes it to blink yellow).

```
sudo dfu-util -d 2b04:d006 -a 0 -s 0x08020000 -D nuttx.bin
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 2b04:d006
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 4096
DfuSe interface name: "Internal Flash   "
Downloading to address = 0x08020000, size = 345560
Download	[=========================] 100%       345560 bytes
Download done.
File downloaded successfully
```

acassis, 2022-09-01:
Hi @greg5678 sorry I only saw your issue today. Did you search to know what the yellow blinking means?
Last time I tested Particle Photon was with NuttX 8.x, I need to test it again with new version.

greg5678, 2022-09-01:
Yes, so it blinking yellow means it's in DFU mode. I made a couple of posts about it on Particle's forum, and they recommend I downgrade it to 0.5.5, but that didn't help. 

https://community.particle.io/t/unable-to-flash-nuttx-to-my-photon/63058/6
https://community.particle.io/t/how-to-downgrade-a-particle-photons-device-os-version/63062/4

greg5678, 2022-09-06:
Any idea what might be the issue?

acassis, 2022-09-10:
Hi @greg5678 I did a little investigation today and it seems that some modification break the network support:
```
ABCDF
netdev_register: Registered MAC: 00:00:00:00:00:00 as dev: wlan0
netdev_ifr_ioctl: cmd: 1794
netdev_ifr_ioctl: cmd: 1796
netdev_ifr_ioctl: cmd: 1800
netdev_ifr_ioctl: cmd: 1818
nsh_telnetstart: Starting the Telnet daemon
telnetd_daemon: Accepting connections on port 23
```

You can see it if you enable: Device Drivers -> Serial Driver Support -> Serial Console = USART1 and Build Setup -> Debug Options -> Enable Debug Features (En. Error, Warn and Info) and Network Debug Features (Error, Warn and Info).

Then use a USB/Serial 3.3V connected to TX/RX/GND pins of the board and use a serial console terminal (i.e. minicom) configured to 115200.

If you download some previous NuttX version you will see it was working, I suspect some recent modification on network generated this issue.

acassis, 2022-09-10:
Ok, I found an old version that worked fine for me. It was from 2018! For some reason DHCP didn't work with my router, then I setup it fixed to 192.168.0.16.
You can find my saved defconfig here: https://pastebin.com/raw/V33BnMVv
If you want to test the binary, just flash it on your board and configure a wifi router (or phone with hotspot) with SSID: DLINK and password: dlinkphoto
More info here: https://acassis.wordpress.com/2022/09/10/nuttx-binary-to-test-particle-photon-board/

```
$ telnet 192.168.0.16
Trying 192.168.0.16...
Connected to 192.168.0.16.
Escape character is '^]'.

NuttShell (NSH)
nsh> uname
NuttX
nsh> ?
help usage:  help [-v] [<cmd>]

  [           cmp         free        mb          pwd         time        
  ?           dirname     help        mkdir       rm          true        
  arp         dd          hexdump     mh          rmdir       uname       
  basename    df          ifconfig    mount       set         umount      
  break       echo        ifdown      mv          sh          unset       
  cat         exec        ifup        mw          sleep       usleep      
  cd          exit        kill        nslookup    test        xd          
  cp          false       ls          ps          telnetd     

Builtin Apps:
  wapi
nsh> 
```

greg5678, 2022-09-13:
So, I downloaded nuttx 8.2 and did the same thing as before, but the device just keep blinking yellow and when I pressed the reset button the device's status LED stayed dark like before. I then took the firmware you gave me and it gave me the same results.

```
sudo dfu-util -d 2b04:d006 -a 0 -s 0x08020000 -D nuttx.bin
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 2b04:d006
Run-time device DFU version 011a
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 011a
Device returned transfer size 4096
DfuSe interface name: "Internal Flash   "
Downloading to address = 0x08020000, size = 336020
Download	[=========================] 100%       336020 bytes
Download done.
File downloaded successfully
```

greg5678, 2022-09-13:
So, since the network support is broken on the newer versions of Nuttx will that get patched eventually?

acassis, 2023-03-19:
@greg5678 I missed your message, probably because other priority tasks. Please refer to this issue: https://github.com/apache/nuttx/issues/8839 I got NuttX 9.0 working on Photon board. Maybe you could help too to find each patch broke it.

BTW, I will close this issue because it is duplicated now.

closed 2023-03-19 by acassis