apache/nuttx #6941: Wrong calculations by pow()

fjpanag, 2022-08-27:
I just realized that `pow()` produces wrong results.

Specifically:

`pow(-2, 2)` yields `inf` instead of 4.

fjpanag, 2022-08-27:
This is the NuttX implementation of `pow()`:

```c
double pow(double b, double e)
{
  return exp(e * log(b));
}
```

A friend of mine advised me that it is incorrect to use `log()` with negative numbers.
*(I am not capable of evaluating such math expressions myself, sorry...)*

So we need special handling for negative bases, I guess?

fjpanag, 2022-08-29:
*Note to self.*

This is how the C standard defines the `pow()` function:

> The pow function computes x raised to the power y. A domain error occurs if x is negative
and y is not an integral value. A domain error occurs if the result cannot be represented when x
is zero and y is less than or equal to zero. A range error may occur.

acassis, 2022-08-29:
@fjpanag I talked with project maintainer of the LibmCS and he trying to get authorization from ESA to let's include it on NuttX to replace our Rhombus' libm implemantation. More info: https://www.youtube.com/watch?v=UcRefVlVL_A

fjpanag, 2022-08-29:
Oh nice! It would be great to include a high quality library in NuttX!

For the record, I fixed this specific function. I just haven't pushed it yet.

I haven't done so because the exact same Lua test fails again, although the function seems to behave correctly now.  
Thus, either my fix is wrong, or other libm functions are called for the same operation, and there is another bug elsewhere...

Compiling with the toolchain provided library fixes all issues.

nealef, 2022-09-02:
Just to add to the libm discussion and motivations for libmCS: https://github.com/apache/incubator-nuttx/issues/6767

fjpanag, 2022-09-02:
So my fix is indeed correct algorithmically, but the library itself is not very accurate. This is what makes the tests fail.

For example, `2^3` yields `8.0000000000000089` instead of just `8.0`.

My toolchain's library tends to be more accurate, and thus it makes the tests pass.

I will provide a PR to fix the issue, so negative bases will be able to be used with `pow()`, but a new more accurate library is surely needed.

closed 2022-09-03 by xiaoxiang781216