apache/nuttx #7008: ReleaseNotes: Cannot link directly to PR URL

baggio63446333, 2022-09-05:
In the ReleaseNotes, e.g. the correct link to a pull request is not
`https://github.com/apache/incubator-nuttx/pull/#5978`
but
`https://github.com/apache/incubator-nuttx/pull/5978`, so no `#` is needed.
I am inconvenienced because I can not link the PR directly from the release notes.
Is it just me? Any particular reason for this?

xiaoxiang781216, 2022-09-06:
@baggio63446333 could you provide a patch fix it?

baggio63446333, 2022-09-06:
I added a pull request to fix a ReleaseNotes on GitHub.
 
I hope the other release pages and Confluence will be fixed as well.
https://nuttx.incubator.apache.org/releases/10.2.0/
https://nuttx.incubator.apache.org/releases/10.3.0/
https://cwiki.apache.org/confluence/display/NUTTX/NuttX+10.2
https://cwiki.apache.org/confluence/display/NUTTX/NuttX+10.3
https://cwiki.apache.org/confluence/display/NUTTX/NuttX+11.0.0

jerpelea, 2022-09-06:
@baggio63446333 

Thanks for notifying us !
I fixed the confluence release notes

closed 2022-09-06 by jerpelea